BrushlessPower/SBUS2-Telemetry #7: Issue getting it to work

emilverwoerd, 2019-10-01:
This is probably not an issue with the library but I'm using an Uno with an SN74HC IC and it's not sending any telemetry data. Maybe there is a way to conversate about a solution.

BrushlessPower, 2019-10-02:
which SN74HC do you use? and which Voltage for the SN74HC?
on my Designs i use SN74LVC2G86 (EXOR) and that's working fine.

The UART on UNO Boards is connected to a Atmega16 as a USB UART Bridge. It should work, but its never tested....see #6 

You could try to remove the Resistor RN4A on the UNO (in TX Line).

Could you try to open the Serial of your UNO Board with 100.000 Baud? You should be able to See the SBUS Data.

Did you have an Oscilloscope?

emilverwoerd, 2019-10-02:
Hi There,

First thx for the quick response. I haven't been completely clear about the
board its an UNO Rev3 so it is the Atmega32 chipset. I also need to say I'm
a software developer and my skills as
an electronic engineer are like almost 0 so it could be that I say
something stupid due to the fact I haven't a lot of knowledge.

The goal of my project is to replace my current S32 Jlog by my own unit
since the jlog doesn't work perfect anymore and the company is not working
on new stuff. I now have a Scorpion Tribunus ESC
which sends telemetry data and I want to parse that to SBUS2 data stream
which I can look at at my T18SZ. I also have the protocol definition of the
ESC from the developer....

The problem I'm facing now not able to send data.

I have changed my 'SN74HC14N' ic to 1 BC547 NPN which allowed me to receive
data from the receiver. Since I have no secondary USB interface yet to
display data on screen. I delayed the on-board let blinking based on
transmitter channel input and hey it worked.

Second step now was the expand the board with a second BC547 NPN. which I
hope have done correctly. I send the data but it is not coming in my
transmitter T18SZ. I also am used to the fact you need to register sensors
and I'm not able to do that just manually add them like people do with the
FBL Brain unit and futaba telemetry which should be fine.

The problem now is that I don't know if my wiring on my board is correct
for SBUS2 retrieval or that it is something else. I also don\t have a temp
sensor on my 18SZ other than F1713.

So maybe you could say if the IC I had was incorrect probably I need to
order some new ones do you have a good idea what I can order from the
Netherlands.

PS: I don't have an oscilioscope yet....

Kind Regards,
Emil Verwoerd

BrushlessPower, 2019-10-07:
Dear Emil,

i think there are some Points to talk about:

Your Board is an Arduino Uno Rev 3:
So there should be an Atmega16 as USB UART Bridge an a Atmega 328P.
I belive the 328P should work with my Library because the TX line of Atmega16 is coupled with a 10k (RN4A)

The SN74HC14N should be fine.
If you are getting SBUS(2) Frames, you should be able to send Telemetry Data.
Please post you schematic of your Hardware Inverter. And check it 3 times.

I think your main Problem is Setting up your T18SZ.

Normally (with Futaba Sensors, JLOG, Unilog ....) you have to connect the sensors with your T18SZ an resgister them. Then the T18SZ tell every single Sensor his SLOT ID.
This will not work with my Library.

You have to set the SLOT Numbers manually! Without that you will not receive anything
I can not tell you how to do i with a T18SZ. I just have a T14SG.
You have to search the setting were all SLOT ID's are listed. Then you have to set the Right Sensor on every SLOT ID...
So following my example:
Set SLOT 1 to Temp Sensor 125
Set SLOT 2 to RPM Sensor
Set SLOT 3 to Current Sensor
and so on....
maybe you can post some pictures from the T18SZ with the workflow.
I think i will add a little Wiki for the T14SG and T18SZ and all the stuff which is nice2know.

If you have no Tempsensor 125 leave SLOT 1 unused. But i think Temp 125 is the same like F1713.
just try it....the worst thing could be a wrong Temperature displayed.

If it's easier for you, write me an Email to brushlesspower@protonmail.com
you can write on German if its easier for you.

emilverwoerd, 2019-10-08:
It seems that the SN74HC14N isn't supported for this application (does not support TTL see wikipedia). I used 2 BC547 NPN resistors which makes it more difficult on the bread board but gives data.

I have tested it now succesfull on my T18SZ Transmitter with the following receivers:
Futaba R7006SB
Futaba CGY 760 (built-in receiver)

closed 2019-10-08 by emilverwoerd