BrushlessPower/SBUS2-Telemetry #10: Radio: Futaba 7PX

wienne, 2019-12-26:
Radio: Futaba 7PX
Version: 4.0S
Receiver: R304SB (T-FHSS)

Currently I only need voltage data.

The value displayed on the remote control is not the voltage value directly specified in the code.

This proportional relationship is puzzling.

[image]

wienne, 2019-12-26:
Value | voltage
-- | --
1 | 0
2 | 0
3 | 0
4 | 0
5 | 2.5
6 | 2.5
7 | 2.5
8 | 2.5
9 | 2.5
10 | 2.5
11 | 2.5
12 | 5.1
13 | 5.1
14 | 5.1
15 | 5.1
16 | 5.1
17 | 5.1
18 | 7.6
19 | 7.6
20 | 7.6
21 | 7.6
22 | 7.6
23 | 7.6
24 | 10.2
25 | 10.2
26 | 10.2
27 | 10.2
28 | 10.2
29 | 10.2
30 | 12.8
31 | 12.8
32 | 12.8
33 | 12.8
34 | 12.8
35 | 12.8
36 | 15.3
37 | 15.3
38 | 15.3
39 | 15.3
40 | 15.3
41 | 15.3
42 | 17.9
43 | 17.9
44 | 17.9
45 | 17.9
46 | 17.9
47 | 17.9
48 | 20.4
49 | 20.4
50 | 20.4
51 | 20.4
52 | 20.4
53 | 20.4
54 | 23
55 | 23
56 | 23
57 | 23
58 | 23
59 | 23
60 | 25.6
61 | 25.6
62 | 25.6
63 | 25.6
64 | 25.6
65 | 25.6
66 | 28.1
67 | 28.1
68 | 28.1
69 | 28.1
70 | 28.1
71 | 28.1
72 | 30.7
73 | 30.7
74 | 30.7
75 | 30.7
76 | 30.7
77 | 30.7
78 | 33.2
79 | 33.2
80 | 33.2
81 | 33.2
82 | 33.2
83 | 33.2
84 | 35.8
85 | 35.8
86 | 35.8
87 | 35.8
88 | 35.8
89 | 35.8
90 | 38.4
91 | 38.4
92 | 38.4
93 | 38.4
94 | 38.4
95 | 38.4
96 | 40.9
97 | 40.9
98 | 40.9
99 | 40.9
100 | 40.9
101 | 40.9
102 | 43.5
103 | 43.5
104 | 43.5
105 | 43.5
106 | 43.5
107 | 43.5
108 | 46
109 | 46
110 | 46
111 | 46
112 | 46
113 | 46
114 | 48.6
115 | 48.6
116 | 48.6
117 | 48.6
118 | 48.6
119 | 48.6
120 | 51.2
121 | 51.2
122 | 51.2
123 | 51.2
124 | 51.2
125 | 51.2
126 | 53.7
127 | 53.7
128 | 53.7
129 | 53.7
130 | 53.7
131 | 53.7
132 | 56.3
133 | 56.3
134 | 56.3
135 | 56.3
136 | 56.3
137 | 56.3
138 | 58.8
139 | 58.8
140 | 58.8
141 | 58.8
142 | 58.8
143 | 58.8
144 | 61.4
145 | 61.4
146 | 61.4
147 | 61.4
148 | 61.4
149 | 61.4
150 | 64
151 | 64
152 | 64
153 | 64
154 | 64
155 | 64
156 | 66.5
157 | 66.5
158 | 66.5
159 | 66.5
160 | 66.5
161 | 66.5
162 | 69.1
163 | 69.1
164 | 69.1
165 | 69.1
166 | 69.1
167 | 69.1
168 | 71.6
169 | 71.6
170 | 71.6
171 | 71.6
172 | 71.6
173 | 71.6
174 | 74.2
175 | 74.2
176 | 74.2
177 | 74.2
178 | 74.2
179 | 74.2
180 | 76.8
181 | 76.8
182 | 76.8
183 | 76.8
184 | 76.8
185 | 76.8
186 | 79.3
187 | 79.3
188 | 79.3
189 | 79.3
190 | 79.3
191 | 79.3
192 | 81.9
193 | 81.9
194 | 81.9
195 | 81.9
196 | 81.9
197 | 81.9
198 | 84.4
199 | 84.4
200 | 84.4
201 | 84.4
202 | 84.4
203 | 84.4
204 | 87
205 | 87
206 | 87
207 | 87
208 | 87
209 | 87
210 | 89.6
211 | 89.6
212 | 89.6
213 | 89.6
214 | 89.6
215 | 89.6
216 | 92.1
217 | 92.1
218 | 92.1
219 | 92.1
220 | 92.1
221 | 92.1
222 | 94.7
223 | 94.7
224 | 94.7
225 | 94.7
226 | 94.7
227 | 94.7
228 | 97.2
229 | 97.2
230 | 97.2
231 | 97.2
232 | 97.2
233 | 97.2
234 | 99.8
235 | 99.8
236 | 99.8
237 | 99.8
238 | 99.8
239 | 99.8
240 | 2.4
241 | 2.4
242 | 2.4
243 | 2.4
244 | 2.4
245 | 2.4
246 | 4.9
247 | 4.9
248 | 4.9
249 | 4.9
250 | 4.9
251 | 4.9
252 | 7.5
253 | 7.9
254 | 7.9
255 | 7.9
256 | 7.9
257 | 7.9
258 | 10
259 | 10
260 | 10
261 | 10
262 | 10
263 | 10
264 | 12.6
265 | 12.6
266 | 12.6
267 | 12.6
268 | 12.6
269 | 12.6
270 | 15.2
271 | 15.2
272 | 15.2
273 | 15.2
274 | 15.2
275 | 15.2
276 | 17.7
277 | 17.7
278 | 17.7
279 | 17.7
280 | 17.7
281 | 17.7
282 | 20.3
283 | 20.3
284 | 20.3
285 | 20.3
286 | 20.3
287 | 20.3
288 | 22.8
289 | 22.8
290 | 22.8
291 | 22.8
292 | 22.8
293 | 22.8
294 | 25.4
295 | 25.4
296 | 25.4
297 | 25.4
298 | 25.4
299 | 25.4
300 | 28
301 | 28
302 | 28
303 | 28
304 | 28
305 | 28
306 | 30.5
307 | 30.5
308 | 30.5
309 | 30.5
310 | 30.5
311 | 30.5
312 | 33.1
313 | 33.1
314 | 33.1
315 | 33.1
316 | 33.1
317 | 33.1
318 | 35.6
319 | 35.6
320 | 35.6
321 | 35.6
322 | 35.6
323 | 35.6
324 | 38.2
325 | 38.2
326 | 38.2
327 | 38.2
328 | 38.2
329 | 38.2
330 | 40.8
331 | 40.8
332 | 40.8
333 | 40.8
334 | 40.8
335 | 40.8
336 | 43.3
337 | 43.3
338 | 43.3
339 | 43.3
340 | 43.3
341 | 43.3
342 | 45.9
343 | 45.9
344 | 45.9
345 | 45.9
346 | 45.9
347 | 45.9
348 | 48.4
349 | 48.4
350 | 48.4
351 | 48.4
352 | 48.4
353 | 48.4
354 | 51
355 | 51
356 | 51
357 | 51
358 | 51
359 | 51
360 | 53.6
361 | 53.6
362 | 53.6
363 | 53.6
364 | 53.6
365 | 53.6
366 | 56.1
367 | 56.1
368 | 56.1
369 | 56.1
370 | 56.1
371 | 56.1
372 | 58.7
373 | 58.7
374 | 58.7
375 | 58.7
376 | 58.7
377 | 58.7
378 | 61.2
379 | 61.2
380 | 61.2
381 | 61.2
382 | 61.2
383 | 61.2
384 | 63.8
385 | 63.8
386 | 63.8
387 | 63.8
388 | 63.8
389 | 63.8
390 | 66.4
391 | 66.4
392 | 66.4
393 | 66.4
394 | 66.4
395 | 66.4
396 | 68.9
397 | 68.9
398 | 68.9
399 | 68.9
400 | 68.9
401 | 68.9
402 | 71.5
403 | 71.5
404 | 71.5
405 | 71.5
406 | 71.5
407 | 71.5
408 | 74
409 | 74
410 | 74
411 | 74
412 | 74
413 | 74
414 | 76.6
415 | 76.6
416 | 76.6
417 | 76.6
418 | 76.6
419 | 76.6
420 | 79.2
421 | 79.2
422 | 79.2
423 | 79.2
424 | 79.2
425 | 79.2
426 | 81.7
427 | 81.7
428 | 81.7
429 | 81.7
430 | 81.7
431 | 81.7
432 | 84.3
433 | 84.3
434 | 84.3
435 | 84.3
436 | 84.3
437 | 84.3
438 | 86.8
439 | 86.8
440 | 86.8
441 | 86.8
442 | 86.8
443 | 86.8
444 | 89.4
445 | 89.4
446 | 89.4
447 | 89.4
448 | 89.4
449 | 89.4
450 | 92
451 | 92
452 | 92
453 | 92
454 | 92
455 | 92
456 | 94.5
457 | 94.5
458 | 94.5
459 | 94.5
460 | 94.5
461 | 94.5
462 | 97.1
463 | 97.1
464 | 97.1
465 | 97.1
466 | 97.1
467 | 97.1
468 | 99.6
469 | 99.6
470 | 99.6
471 | 99.6
472 | 99.6
473 | 99.6
474 | 2.2
475 | 2.2
476 | 2.2
477 | 2.2
478 | 2.2
479 | 2.2
480 | 4.8
481 | 4.8
482 | 4.8
483 | 4.8
484 | 4.8
485 | 4.8
486 | 7.3
487 | 7.3
488 | 7.3
489 | 7.3
490 | 7.3
491 | 7.3
492 | 9.9
493 | 9.9
494 | 9.9
495 | 9.9
496 | 9.9
497 | 9.9
498 | 12.4
499 | 12.4
500 | 12.4
501 | 12.4
502 | 12.4
503 | 12.4
504 | 15
505 | 15
506 | 15
507 | 15
508 | 15
509 | 15
510 | 17.6

wienne, 2019-12-26:
[image]

BrushlessPower, 2020-01-06:
Hello Wienne,

could you give me which Sensor you use in your 7PX?
SBS/01V?
F1678?
These two Sensors have different internal Logic. The F1678 (Current,Voltage,Capacity) is Supported.
SBS/01V is actual not supported. But i could implement this Sensor.

are the other Sensors working?

Greetings

BrushlessPower, 2020-01-06:
just saw in the Futaba Documntation that the F1678 is not supported by the 7PX

is that right?
so i assume that you are using the SBS/01V Sensor.

Give me some Days, i will send you some code.
If its working i will add it to the Code on Git

....maybe i should add Sensors with the real Name to avoid misunderstandings

wienne, 2020-01-06:
Thank you very much for replying to this post.

I use a current sensor. Only this sensor has a change in voltage.

Others, such as temperature, also change, but the values ​​are too exaggerated.
For example, the test value is several tens, but the displayed temperature is several thousands.
RPM speed, GPS, there is a return value, but the value is absolutely incorrect.
It's just that I don't plan to use them at the moment, so I didn't pay attention to them.

A name like F1678 confuses me a bit,
I seem to see many similar names in the 18SZ sensor list.
I didn't know their role, so I didn't pay attention to and remember their names.
It is impossible to determine whether the name of the sensor you mentioned is used.

Until this morning, I tested the same code using Futaba 18sz and 7008SB.
It was just a little rushed in the morning and I simply tested the current sensor.
Like 7PX, only the voltage, current, and capacity in the current sensor have numerical changes, and the rules are the same.
The separate voltage sensor does not return any value.

The current in the current sensor has a more intuitive proportional relationship with the test value.
For example, 100 will show a current of 1A. 800-> 8A
However, 1000 will also show a current of 1A, 1500-> 15A.
Detailed and accurate correspondence, I have to do testing and recording.

Due to a business trip today, I will not be able to return home until January 11, 2020.
After relevant tests,
I can then reply to the exact names of the sensors I use, and which sensors will display changing values, and the law between the values ​​and the voltage.

 Your response gives me the motivation to perform these tests, thank you very much.

BrushlessPower, 2020-01-06:
dear Wienne,

for further support, please send me your arduino sketch. or use the default sketch from github.

next general thing:
a current value of 100 is 1.00 A in the display of your 7PX or 18SZ.
So the real current has to be multiplied by 100
the same for Voltage. A Value of 1234 is displayed as 12.34V

Please look at the comments in my example sketch.

all other issues could be caused by wrong Sensor settings
but i need the exact settings in your 7PX and 18SZ.
the 18SZ with R7008 should run all Sensors without issues.

until 11, January i will send you the code (with instructions) for the Voltage-Only-Sensor (SBS/01V)

if it's easier for you, you can write me an email: brushlesspower@protonmail.com

BrushlessPower, 2020-01-06:
here is the Code for you:
its the implementation of the SBS/01V Voltage Sensor. This Sensor measures 2 Voltages. RX Voltage and external Voltage. It's the same like the Voltage measuring done in R7008.

.INO file:
#define VOLTAGE_SLOT      16    // 2 Slot Sensor

in loop add:
send_voltage(VOLTAGE_SLOT,(uint16_t)128, (uint16_t)255);                                                        // Voltage1 = 12.8V, Voltage2 = 25.5V

in SBUS2.cpp add:
void send_voltage(uint8_t port,uint16_t voltage1, uint16_t voltage2)
{
   uint16_t value = 0;
   uint8_t bytes[SBUS2_TEL_DATA_SIZE] = {0x03, 0x40, 0x00 };
   
   // VOLTAGE1
   value = voltage1 | 0x8000; 
   if ( value > 0x9FFF )
   {
      // max voltage is 819.1
      value = 0x9FFF;
   }  
   bytes[1] = value >> 8;
   bytes[2] = value;
   SBUS2_transmit_telemetry_data( port , bytes);

   //VOLTAGE2
   value = voltage2;
   if ( value > 0x1FFF )
   {
      // max voltagee is 819.1
      value = 0x1FFF;
   }  
   bytes[1] = value >> 8;
   bytes[2] = value;
   SBUS2_transmit_telemetry_data( port+1 , bytes);
}

in SBUS2.h add:
void send_voltage(uint8_t port,uint16_t voltage1, uint16_t voltage2);

wienne, 2020-01-10:
Setting the slot where the sensor is located is key.

Temp-F1713 temperature sensor must be in slot 1 or 8 or 9.

The current sensor or F1678 sensor can only be used in slot 3.

The RPM sensor can only be used in slot 2.
The specified value is slightly different from the displayed value.

GPS-F1675 sensor, only in slot 8.

Voltage sensor only in slot 16.
In slot 10, the battery voltage value is incorrect, and the ratio of the external voltage value to the display is 1: 1
At 16 slots, the value of the battery voltage is the same as the value of the external voltage, and the ratio of the specified value to the displayed value is 10: 1, which is the same as the example in the code.

The above test was successful in 18SZ + R7008SB.

At 7PX + R304SB, except for GPS, it is the same as 18SZ.
There is no GPS-F1675 sensor in the 7PX, only the GPS sensor.
In any slot (8,16,24) is set to GPS, the value is empty.

I thought of another question, if there are more than two temperature sensors, how should I achieve it?
Using more than two arduino?

Is there a code for the altitude sensor currently?

ericlangel, 2020-01-13:
Dear Wienne,

its hard to understand you correctly. I think you dont know much about Futaba Telemetry?
And i think you dont read any Comments or documentation.

"Temp-F1713 temperature sensor must be in slot 1 or 8 or 9."
-> NO. With my default Arduino project you have to Set the Sensor to SLOT 1. Slot 8 and 9 are for GPS Module.

"In slot 10, the battery voltage value is incorrect, and the ratio of the external voltage value to the display is 1: 1"
-> Of course it's incorrect. SLOT 10 is GPS Sensor

"There is no GPS-F1675 sensor in the 7PX, only the GPS sensor.
In any slot (8,16,24) is set to GPS, the value is empty."
-> Right. The 7PX doesn't support the F1675. And there you have a Problem: The F1675 Sensor is the only documented GPS Sensor.
All other GPS Sensors are not documented. So i cant implement them to this Library.

"I thought of another question, if there are more than two temperature sensors, how should I achieve it?
Using more than two arduino?"
-> No, just change the Slot Number:
send_temp125(1, (int16_t)50);  -> Slot 1
send_temp125(2, (int16_t)60);  -> Slot 2   
send_temp125(3, (int16_t)70);  -> Slot 3  
send_temp125(4, (int16_t)80);  -> Slot 4

Keep in Mind....just 1 Sensor per Slot!
If you have more Problems, please send me pictures of the 7PX Display and your Arduino Project. 

"Is there a code for the altitude sensor currently?"
-> No. it's planned with the next release

BrushlessPower, 2020-04-07:
i think the Problems are solved with the new Readme

closed 2020-04-07 by BrushlessPower