BrushlessPower/SBUS2-Telemetry #12: GPS Latitude & Longitude problem

magnum3131, 2020-01-08:
Hi,
Thanks for all your work, 

I have a problem with negative GPS coordinates for South and West, it's working fine with positive (North and East).

If i use the coordinate
int32_t latitude = -52312499;    // =S 52° 31.2499 = N 52.520833 = N 52° 31' 14.9988"
int32_t longitude = -13245658;   // = W 13° 24.5658 = E 13.409430 = E 13° 24' 33.9480"

The radio display S 205°13.6077, W 244°20.2918

Any idea ?

Thanks
Julien

BrushlessPower, 2020-01-09:
thank you for posting your issue.
i never tested negative coordinates.

could you replace your "send_f1675_gps" function in sbus2.cpp with following code:

void send_f1675_gps(uint8_t port, uint16_t speed, int16_t altitude, int16_t vario, int32_t latitude, int32_t longitude)
{
   uint16_t value1 = 0;
   int16_t  value2 = 0;
   uint32_t  value3 = 0;
   bool latitude_pos = false;
   bool longitude_pos = false;
   uint8_t bytes[SBUS2_TEL_DATA_SIZE] = {0x03, 0x40, 0x00 };
 
   
   // SPEED -> Bit 14(bytes[1] bit7) -> GPS Valid or not
   value1 = speed | 0x4000;
   if ( value1 > 0x43E7 )
   {
      // max Speed is 999 km/h
      value1 = 0x43E7;
   }  
   bytes[1] = value1 >> 8;
   bytes[2] = value1;
   SBUS2_transmit_telemetry_data( port , bytes);

   //HIGHT
   value2 = altitude | 0x4000;
   bytes[1] = value2 >> 8;
   bytes[2] = value2;
   SBUS2_transmit_telemetry_data( port+1 , bytes);

   //TIME -> 12:34:56 Uhr = 12*60*60 + 34*60 + 56 = 45296 = 0xB0F0
   bytes[1] = 0xB0;
   bytes[2] = 0xF0;
   SBUS2_transmit_telemetry_data( port+2 , bytes);

   // VARIO
   value2 = vario * 10;   
   bytes[1] = value2 >> 8;
   bytes[2] = value2;
   SBUS2_transmit_telemetry_data( port+3 , bytes);

   // LATITUDE
   if(latitude >= 0){
    latitude_pos = true;
   }
   else{
    latitude_pos = false;
    latitude = latitude * -1;
   }
   bytes[1] = (uint8_t)(latitude/1000000);
   value3 = (uint32_t)(latitude%1000000);
   if(latitude_pos){
    bytes[2] = ((value3 >> 16) & 0x0f);    // North
   }
   else{
    bytes[2] = ((value3 >> 16) & 0x1f);    // South
   }
   SBUS2_transmit_telemetry_data( port+4 , bytes);

   bytes[1] = ((value3 >> 8) & 0xff);
   bytes[2] = value3 & 0xff;
   SBUS2_transmit_telemetry_data( port+5 , bytes);

   // LONGITUDE
   if(longitude >= 0){
    longitude_pos = true;
   }
   else{
    longitude_pos = false;
    longitude = longitude * -1;
   }
   bytes[1] = (uint8_t)(longitude/1000000);
   value3 = (uint32_t)(longitude%1000000);
   if(longitude_pos){
    bytes[2] = ((value3 >> 16) & 0x0f);    // Eath
   }
   else{
    bytes[2] = ((value3 >> 16) & 0x1f);    // West
   }
   SBUS2_transmit_telemetry_data( port+6 , bytes);

   bytes[1] = ((value3 >> 8) & 0xff);
   bytes[2] = value3 & 0xff;
   SBUS2_transmit_telemetry_data( port+7 , bytes);
}

BrushlessPower, 2020-01-09:
if it's working i will change it in Git

magnum3131, 2020-01-09:
Hi thanks for your answer,

The problem is different now, in both case (positive or negative cordinate) the radio always display
North for Latitude and East for longitude. The coordinate are good, there is just a problem of pole

Exemple
int32_t latitude = -52312499; // Display = N 52° 31.2499
int32_t longitude = -13245658; // Display = E 13° 24.5658

Thanks
Julien

ericlangel, 2020-01-13:
Dear Julien,

i'am not sure why the Pole is wrong. At the Moment i have not the Time to Debug the Problem on my T14SG.
Maybe you have to wait until i have some more time.

please try the following code:

void send_f1675_gps(uint8_t port, uint16_t speed, int16_t altitude, int16_t vario, int32_t latitude, int32_t longitude)
{
   uint16_t value1 = 0;
   int16_t  value2 = 0;
   uint32_t  value3 = 0;
   bool latitude_pos = false;
   bool longitude_pos = false;
   int32_t _latitude = latitude;
   int32_t _longitude = longitude;
   uint8_t bytes[SBUS2_TEL_DATA_SIZE] = {0x03, 0x40, 0x00 };
 
   
   // SPEED -> Bit 14(bytes[1] bit7) -> GPS Valid or not
   value1 = speed | 0x4000;
   if ( value1 > 0x43E7 )
   {
      // max Speed is 999 km/h
      value1 = 0x43E7;
   }  
   bytes[1] = value1 >> 8;
   bytes[2] = value1;
   SBUS2_transmit_telemetry_data( port , bytes);

   //HIGHT
   value2 = altitude | 0x4000;
   bytes[1] = value2 >> 8;
   bytes[2] = value2;
   SBUS2_transmit_telemetry_data( port+1 , bytes);

   //TIME -> 12:34:56 Uhr = 12*60*60 + 34*60 + 56 = 45296 = 0xB0F0
   bytes[1] = 0xB0;
   bytes[2] = 0xF0;
   SBUS2_transmit_telemetry_data( port+2 , bytes);

   // VARIO
   value2 = vario * 10;   
   bytes[1] = value2 >> 8;
   bytes[2] = value2;
   SBUS2_transmit_telemetry_data( port+3 , bytes);

   // LATITUDE
   if(latitude >= 0){
    latitude_pos = true;
   }
   else{
    latitude_pos = false;
    latitude = latitude * -1;
   }
   bytes[1] = (uint8_t)(latitude/1000000);
   value3 = (uint32_t)(latitude%1000000);
   if(_latitude >= 0){
    bytes[2] = ((value3 >> 16) & 0x0f);    // North
   }
   else{
    bytes[2] = ((value3 >> 16) & 0x1f);    // South
   }
   SBUS2_transmit_telemetry_data( port+4 , bytes);

   bytes[1] = ((value3 >> 8) & 0xff);
   bytes[2] = value3 & 0xff;
   SBUS2_transmit_telemetry_data( port+5 , bytes);

   // LONGITUDE
   if(longitude >= 0){
    longitude_pos = true;
   }
   else{
    longitude_pos = false;
    longitude = longitude * -1;
   }
   bytes[1] = (uint8_t)(longitude/1000000);
   value3 = (uint32_t)(longitude%1000000);
   if(_longitude >= 0){
    bytes[2] = ((value3 >> 16) & 0x0f);    // Eath
   }
   else{
    bytes[2] = ((value3 >> 16) & 0x1f);    // West
   }
   SBUS2_transmit_telemetry_data( port+6 , bytes);

   bytes[1] = ((value3 >> 8) & 0xff);
   bytes[2] = value3 & 0xff;
   SBUS2_transmit_telemetry_data( port+7 , bytes);
}

magnum3131, 2020-01-13:
Hi,
Still the same problem with your code, i'll try to working on it too
Thanks a lot
Julien

ericlangel, 2020-01-13:
the GPS Sensor is documented here:
https://sites.google.com/site/sbus2diy/home/sbus2protocol-part2/sbus2protocol-part3

please use google translation

North/south and East/West is just a single Bit set to 1 or set to zero.

you could try to send the Longitude value without calculation. It's just shifting some Bits

// LONGITUDE 13245658 -> 13° and 245658 = 0x 3 BF 9A
bytes[1] = 0x0D; -> 13 Degree
bytes[2] = 0x03; // East = 0
or
bytes[2] = 0x13; // West = 1

SBUS2_transmit_telemetry_data( port+6 , bytes);

bytes[1] = 0xBF;
bytes[2] = 0x9A;
SBUS2_transmit_telemetry_data( port+7 , bytes);
}

magnum3131, 2020-01-13:
Yes it's working without calculation  

bytes[2] = 0x13; // West = 1
I'll do some more test this afternoon

Thanks a lot

Julien

magnum3131, 2020-01-13:
Hi,
I think i found the problem

For the calculation of negative we have to do a bitwise with a XOR and not a AND
bytes[2] = ((value3 >> 16) & 0x1f);    => bytes[2] = ((value3 >> 16)^ 0x10);

It's working with 
int32_t latitude = -52312499; // Display = S 52° 31.2499
int32_t longitude = -13245658; // Display = W 13° 24.5658

Thanks
Julien

ericlangel, 2020-01-13:
on the first view it's sound wrong to me.
did you try it with OR?

245658 >> 16 = 0b0011 = 0x03
(0x03 & 0x1f) = 0001 0011

0x03 ^ 0x10 = 0001 0011

0x03 | 0x10 = 0001 0011

The Result's are the same

magnum3131, 2020-01-13:
Hi
(0x03 & 0x1f) = 0000 0011 = 0x03
I'll try with OR
Thanks 
Julien

ericlangel, 2020-01-13:
yes, you are right...now i see the Problem.

But i think OR should be working and is the better solution

BrushlessPower, 2020-04-20:
solved in V1.0 Release

closed 2020-04-20 by BrushlessPower