BrushlessPower/SBUS2-Telemetry #13: Futaba SBUS2 problem

flyingsinver, 2020-02-05:
Hello  BrushlessPower ,

First of all, thank you for your effort and desire to share your project with others.

Excellent work !!!

Following your instructions I am trying to visualize Futaba telemetry parameters for a new big plane project that I am doing. This aircraft has a four-cylinder DLE 222 engine installed and I would like to visualize the four temperatures of the engine cylinder heads and the interior of the plane. ( five temperature sensor totally )

Also the RPMs and the voltage of the four batteries. In principle I do not need GPS although I do not rule out later using it but if necessary I can do without it.

My initial problem is that I can only display the RPMs, neither the temperature sensor, nor the GPS, etc. etc. appears, also when initializing the variometer it indicates +255.5m/s and at the same time that the RPMs are shown the value of the variometer changes to -64.5m/s

[https://drive.google.com/open?id=1qUmKyzaBxOAnVxb1mKAuKPJhlGejBZo3](url)

[https://drive.google.com/open?id=1qUmKyzaBxOAnVxb1mKAuKPJhlGejBZo3](url)

[https://drive.google.com/open?id=1Jl31XbzBUovFmRNBtNpjTzZlcI4S-0z4](url)

I have tried with arduino atmega 328P at 3.3V 8Mhz and also at 5V and 16Mhz but the result is the same.

I don't have GPS connected.

The configuration of Futaba is T18SZ and two receivers R7108SB dual mode and FassTest 18ch

This afternoon I will try a single receiver and other types of modulation that include telemetry and I will report if I have had any progress but in any case I will need to make it work at least in FassTest 12ch and two receivers.

Any guidance or advice of what you think I am doing wrong and for that reason I cannot make it work I will thank you very much.

In any case, thank you very much for your time and help!

Best regards,
Chema

BrushlessPower, 2020-02-05:
Dear Chema,

i think there is a mismatch of your 18SZ Sensor Settings and the Arduino Sensor Settings.

try to keep it simple:

remove all unused Sensors in Arduino and in your 18SZ

Watch out your Slot Settings, they must be the same in arduino and 18SZ

Fur further help i would be good to post a picture with your 18SZ Slot Settings.

The fact, that the RPM is working, shows that your Hardware and Modulation is working. the Rest is just Firmware and Settings.

Please do not use a Variometer. It's not supported yet.

For Temp125 use "F1713" or "SBS/01TE" if there is no temp125 Sensor in your 18SZ

flyingsinver, 2020-02-05:
Thank you very much for your quick response
I think I have understood you
Tonight i will check

I edited the link for pics

https://drive.google.com/file/d/1NBS2yBeSfRPz0MLVWmmkfSwXrp879I96/view

https://drive.google.com/file/d/1qUmKyzaBxOAnVxb1mKAuKPJhlGejBZo3/view

https://drive.google.com/file/d/1Jl31XbzBUovFmRNBtNpjTzZlcI4S-0z4/view

I am very happy to have established communication with you !!!

Thank you very much again

flyingsinver, 2020-02-05:
Niceeeeeeee !!! Thanks thanks thank you very much my friend !!!

Thank you for your help !!!

I've seen in another thread this comment of yours also helping wienne

----------------------------------------------------------------------------------

.INO file:
#define VOLTAGE_SLOT 16 // 2 Slot Sensor

in loop add:
send_voltage(VOLTAGE_SLOT,(uint16_t)128, (uint16_t)255); // Voltage1 = 12.8V, Voltage2 = 25.5V

in SBUS2.cpp add:
void send_voltage(uint8_t port,uint16_t voltage1, uint16_t voltage2)
{
uint16_t value = 0;
uint8_t bytes[SBUS2_TEL_DATA_SIZE] = {0x03, 0x40, 0x00 };

// VOLTAGE1
value = voltage1 | 0x8000;
if ( value > 0x9FFF )
{
// max voltage is 819.1
value = 0x9FFF;
}
bytes[1] = value >> 8;
bytes[2] = value;
SBUS2_transmit_telemetry_data( port , bytes);

//VOLTAGE2
value = voltage2;
if ( value > 0x1FFF )
{
// max voltagee is 819.1
value = 0x1FFF;
}
bytes[1] = value >> 8;
bytes[2] = value;
SBUS2_transmit_telemetry_data( port+1 , bytes);
}

in SBUS2.h add:
void send_voltage(uint8_t port,uint16_t voltage1, uint16_t voltage2);

--------------------------------------------------------------------------------------

I think it is the modification that I also need to add two additional voltage measurements on board my plane and have four voltage measurements in total for the 4 LiPo 2S batteries that I will install since everything will work at 7.4V

In the absence of the configuration of the additional voltages I leave some photos of the menus of Futaba configuring the sensors, I think they are well indeed seems to work correctly

https://drive.google.com/file/d/1Kik4Rcfz_t6tFIAHXhREoIEbcLuXQHgL/view

https://drive.google.com/file/d/1KjgXWEQoix7WyCnaaA4uJ7w-DXlLr5-1/view

https://drive.google.com/file/d/1KjuiLZct-r2aV9DYVtkh8rxrMl2aIDQS/view

https://drive.google.com/file/d/1L4WmTB7ck1o71JlLxAEuTU-G57gnrw0H/view

https://drive.google.com/file/d/1LDZ8cT5g7SGfZ9YT65cBIls0I70FNEmF/view

https://drive.google.com/file/d/1LFEi697SfE1F8m2D7jxqhnlEY2DT7efX/view

also a litte video of a dynamic simulation ( no real hardware data adquisition )

https://www.youtube.com/watch?v=q_-Rb75BchQ

I will continue moving forward with my project thanks to you !!!

Sincerely,
Chema

BrushlessPower, 2020-02-06:
Nice to hear that it's working.

Most Issues are related to wrong Setting. I plan to write a "How to" with the Release of Version 1.0.
In Version 1.0 i will add the Voltage Sensor and the Vario's

Can i user your pictures of the 18SZ Settings for the How to?
Maybe it would be nice to place the Github Link to the Youtube Video.

Did you build your Inverters on your own? Or did you buy something? Because it looks very nice and clean.

Greetings John

flyingsinver, 2020-02-06:
Hello Jonh !!!

_**-Can i user your pictures of the 18SZ Settings for the How to?**_

Of course, it will be a pleasure to contribute a grain of sand although very small to your enormous contribution!

_**Maybe it would be nice to place the Github Link to the Youtube Video.**_

good idea, done !!!

_**Did you build your Inverters on your own? Or did you buy something? Because it looks very nice and clean.**_

I made it but they are the transistors and the resistances of their scheme, this:

https://github.com/BrushlessPower/SBUS2-Telemetry/blob/master/SBUS2_inverter.png

John, when do you plan to publish the new version ?

Could you extract the value of the RSSI signal from the SBUS and reinject it again as a numerical value from 0 to 100 in percentage or in dB?

Could this be possible?

In this way we would know more exactly as the radio signal arrives at our receiver since now there is only a very small graphic in the display at the top left with the coverage stripes symbol as in mobile phones.

Thanks a lot again for everything

Best regards,
Chema

BrushlessPower, 2020-02-07:
> when do you plan to publish the new version ?

difficult to say. Maybe tomorrow, maybe in some month. I need some free time to test the new code and make a Wiki/How to.
But actual i'am very busy with other stuff.

> Could you extract the value of the RSSI signal from the SBUS and reinject it again as a numerical value from 0 to 100 in percentage or in dB?

That's not so easy. Because i cant find any information about the RSSI (FER -> Frame Error Rate) in the SBUS2 Protocol. Its there, but i don't know where. I will try to get some Information's an add this as a own issue.
But at the End, it will be just a value in Code. You have to send this value as a Temperature (Temp125 Sensor) and set up a alarm on your 18SZ if Temperature exceeds the limit.
Because Futaba has no "RSSI Sensor"

Greetings John

flyingsinver, 2020-02-09:
Ok ok I understand John

In the end this kind of thing is done when you can

If the RSSI data can be implemented, could it be displayed as m / s "monitor signal" or in the worst case as V "value"?

I say it because having that data as ºC is not pretty

I continue with my project thanks to you

Now I am trying to visualize the two additional voltages and finally the actual rpms read to the engine's own electronic ignition so as not to have to use an additional sensor

I will keep informing

best regards,
Chema

wienne, 2020-02-10:
Multiple temperature sensor functions can be implemented.

The key is the definition of the slot number in the program code.
As well as the choice of sensor model in the remote control.

The value of the fifth temperature sensor is 0, because my chip wiring is a little problem.
Sometimes I can see the value when I touch the wire lightly.

[image]

[image]

[image]

http://www.egotop.com/~sunjun/sbus2/sbus2-1.jpg
http://www.egotop.com/~sunjun/sbus2/sbus2-2.jpg
http://www.egotop.com/~sunjun/sbus2/sbus2-3.jpg

`
#define TEMP_SLOT_1   1     // 第1个温度传感器，占1个槽位，起始插槽 1-31
#define TEMP_SLOT_2   2     // 第2个温度传感器，占1个槽位
#define TEMP_SLOT_3   3     // 第3个温度传感器，占1个槽位
#define TEMP_SLOT_4   4     // 第4个温度传感器，占1个槽位
#define TEMP_SLOT_5   5     // 第5个温度传感器，占1个槽位
#define ERROR_SLOT    6     // 占1个槽位，用于程序排错，利用温度传感器显示SBUS2的错误代码
#define CURRENT_SLOT  8     // 电流传感器，占3个槽位，其中有电流、电压、容量三种数值
//CURRENT_SLOT      9，电流传感器，在第8槽时，还将占用、9和10槽。
//CURRENT_SLOT      10
//起始插槽，只能为1-5、8-13、16-21、24-29
#define VOLTAGE_SLOT  11    // 2 Slot Sensor 电压传感器，有两个电压值
//起始槽号只能为1-6、8-14、16-22、24-30

`

wienne, 2020-02-10:
`#define TEMP_SLOT_1 1 // The first temperature sensor, occupying one slot, starting slot 1-31
#define TEMP_SLOT_2 2 // 2nd temperature sensor, occupying 1 slot
#define TEMP_SLOT_3 3 // 3rd temperature sensor, occupying 1 slot
#define TEMP_SLOT_4 4 // 4th temperature sensor, occupying 1 slot
#define TEMP_SLOT_5 5 // 5th temperature sensor, occupying 1 slot
#define ERROR_SLOT 6 // Occupies 1 slot, used for program troubleshooting, and uses the temperature sensor to display the error code of SBUS2
#define CURRENT_SLOT 8 // Current sensor, which occupies 3 slots, including current, voltage, and capacity values
// CURRENT_SLOT 9, the current sensor will also occupy slots 9 and 10 in slot 8.
// CURRENT_SLOT 10
// The starting slot can only be 1-5, 8-13, 16-21, 24-29
#define VOLTAGE_SLOT 11 // 2 Slot Sensor voltage sensor with two voltage values
// The starting slot number can only be 1-6, 8-14, 16-22, 24-30`

flyingsinver, 2020-02-10:
hello

ok ok i understand it thank you very much wienne

I just successfully implemented the additional voltages this is starting to look great

a question for both or for the code creator

What exactly is this line for in the send loop?

what would happen if i remove it? Is it totally necessary?

send_alarm_as_temp125(ERROR_SLOT, ((failsave*1000) + (transmision_dropt_frame*100) + uart_dropped_frame));      // Warning with over Temp at Error Slot

thks !!!!

wienne, 2020-02-11:
This line of code, according to my actual test, if deleted, does not affect the use of other sensors.

According to the naming of variables in the code, I guess it is related to errors, such as interference during signal transmission, packet loss, or data related to RSSI.
I hope the author John can explain this in more detail.

BrushlessPower, 2020-02-11:
> send_alarm_as_temp125(ERROR_SLOT, ((failsave1000) + (transmision_dropt_frame100) + uart_dropped_frame)); // Warning with over Temp at Error Slot

This code is just a Temperature Sensor with special Values.
You can remove it.

The Sensor should always Display Zero on your Futaba Screen

if the receiver indicate the failsafe Flag -> your Screen shows 1000°C
if there was a transmission problem -> your Screen shows 100°C
if there was a internal UART decoding Problem -> Your Screen shows 1°C (increasing with every problem)

For Developing & testing it's a nice "Sensor" to show problems with your Hardware

With older Versions of this Library it was a normal thing that some Errors occur. Now its more stable and it really should be 0°C

BrushlessPower, 2020-04-20:
should be solved?!

closed 2020-04-20 by BrushlessPower