BrushlessPower/SBUS2-Telemetry #14: Output FER -> Frame Error Rate

BrushlessPower, 2020-02-07:
SBUS2 Protocol has a Frame Error Rate (RSSI)

There should be a function to get this Value.

http://www.futaba-forum.net/showthread.php?7674-R7008SB-Empfangsst%E4rke-RSSI-verwenden/page2

Maybe its Decoded in Byte 23 Bit 0 to 3

BrushlessPower, 2020-02-20:
i thin i found an explanation for the Frame Error Rate in 0%, 25%, 50%, 75% and 100%

Its a mix of all values showing the quality of the SBUS Signal.

if the transmision_dropt_frame flag is set, or the failsave flag the FER should be 0%, 25%

the other percents are calculated from the SBUS decoding.

if a frame is completely not received -> maybe 50%?
if a received frame has wrong values -> maybe 75%?

if every frame is received without timeout, without error -> 100%

There is absolutely no Information about the FER on the Internet.
So i think i will add my own FER/RSSI function, based on the SBUS2_get_status() function

BrushlessPower, 2020-02-20:
a second Method:

transmision_dropt_frame flag over the last 100 received frames

see: https://github.com/RealTadango/FrSky/blob/master/Sensors/LinkQuality/src/LinkQualitySensor.ino

emilverwoerd, 2020-03-02:
> i thin i found an explanation for the Frame Error Rate in 0%, 25%, 50%, 75% and 100%
> 
> Its a mix of all values showing the quality of the SBUS Signal.
> 
> if the transmision_dropt_frame flag is set, or the failsave flag the FER should be 0%, 25%
> 
> the other percents are calculated from the SBUS decoding.
> 
> if a frame is completely not received (timeout) -> maybe 50%?
> if a received frame has wrong values (Start/End Byte) -> maybe 75%?
> 
> if every frame is received without timeout, without error -> 100%
> 
> There is absolutely no Information about the FER on the Internet.
> So i think i will add my own FER/RSSI function, based on the SBUS2_get_status() function

Shouldn't you read a channel value as it seems in the article the PWM value looks at what the strength is?

BrushlessPower, 2020-03-03:
> Shouldn't you read a channel value as it seems in the article the PWM value looks at what the strength is?

No.
Reading a Channel Value just gives you the PWM Value (Stick Position). But no RRSI

In the SBUS(2) Protocol are bits for Failsafe and Frame Lost. 
But there is discription how to "translate" it in RSSI %

So i think i will implement it the same way as it's done here: https://github.com/RealTadango/FrSky/blob/master/Sensors/LinkQuality/src/LinkQualitySensor.ino

BrushlessPower, 2020-04-20:
added in V1.0 Release

closed 2020-04-20 by BrushlessPower