BrushlessPower/SBUS2-Telemetry #19: GPS Latitude & Longitude Display problem

wienne, 2020-04-06:
Hello there!

I started to study GPS. Currently the data transmission is normal, but the latitude and longitude display of GPS is inconsistent.
Remote control and receiver, 18SZ + R7008SB.

According to the example in your code, only the decimal point of the latitude and longitude in the GPS data is removed.

For example: GPS data format of NMEA01843
N 3029.52639, displayed as N 47 ° 35.2783
E 11425.15639, displayed as E 118 ° 51.5596
The last degree value may be inconsistent with the original GPS data,
The delay in the display prevented me from maintaining complete accuracy between them.

In addition, in the sample program
In the setup () section, why do latitude and longitude increase by 100?
What is the impact of not increasing by 100?

In your and others' information, I also modified the SBUS2.cpp file.
The following code is used:
 
`
  // LATITUDE
  if(latitude >= 0){
    latitude_pos = true;
  }
  else{
    latitude_pos = false;
    latitude = latitude * -1;
  }
  bytes[1] = (uint8_t)(latitude/1000000);
  value3 = (uint32_t)(latitude%1000000);
  if(_latitude >= 0){
    bytes[2] = ((value3 >> 16) & 0x0f); // North
  }
  else{
    bytes[2] = ((value3 >> 16) & 0x10); // South
  }
  SBUS2_transmit_telemetry_data( port+4 , bytes);

  bytes[1] = ((value3 >> 8) & 0xff);
  bytes[2] = value3 & 0xff;
  SBUS2_transmit_telemetry_data( port+5 , bytes);

  // LONGITUDE
  if(longitude >= 0){
    longitude_pos = true;
  }
  else{
    longitude_pos = false;
    longitude = longitude * -1;
  }
  bytes[1] = (uint8_t)(longitude/1000000);
  value3 = (uint32_t)(longitude%1000000);
  if(_longitude >= 0){
    bytes[2] = ((value3 >> 16) & 0x0f); // Eath
  }
  else{
    bytes[2] = ((value3 >> 16) & 0x10); // West
  }
  SBUS2_transmit_telemetry_data( port+6 , bytes);

  bytes[1] = ((value3 >> 8) & 0xff);
  bytes[2] = value3 & 0xff;
  SBUS2_transmit_telemetry_data( port+7 , bytes);

`

BrushlessPower, 2020-04-06:
Dear wienne,

the GPS Sensor is a little bit confusing, because of the input Type.

You have to use "Degree Minutes" without decimal point (interger).

N 3029.52639 -> which Format is it? Decimal degree? Degree Minutes? Degree Minutes Seconds?
E 11425.15639, -> which Format is it? Decimal degree? Degree Minutes? Degree Minutes Seconds?

you need to convert it to Degree Minutes without decimal point

For Version 1.0 i already modified the GPS Code. So you can choose different input Types. But you have to know i which format you have the GPS Data.

Looking @ the example:

// coordinates Berlin TV Tower  = Degree Minutes  = Decimal Degrees = Degree Minutes Seconds
int32_t latitude = 52312499;    // = N 52° 31.2499 = N 52.520833 = N 52° 31' 14.9988"
int32_t longitude = 13245658;   // = E 13° 24.5658 = E 13.409430 = E 13° 24' 33.9480"

This coordinates are exactly the same position, but they are expressed in different formats

https://www.koordinaten-umrechner.de/decimal/30.492106,114.419273?karte=OpenStreetMap&zoom=8

i used this tool to convert the different types. Its in german, but i think it should be clear.

> In the setup () section, why do latitude and longitude increase by 100?
What is the impact of not increasing by 100?

This is just for testing. If you have a real GPS Sensor you won't need it!
Your Futaba Radio show's the Distance. The Distance i calculated from the LAT and LON Values.

So if your are sending the same LAT/LON the Distance will be 0
So i send position x +100 as the fist LAT/LON Position
And in Loop() i send just position x.
So my Futaba Radio shows the Distance between this coordinates

could you write me an Email to brushlesspower@protonmail.com
then i could give you the Code of Version 1.0 (The readme is already written for 1.0, but i have not tested all sensors)
Maybe you could do some tests on GPS

BrushlessPower, 2020-04-06:
Are the coordinates of Berlin displayed correctly?

Which GPS Sensor did you use?
Only GPS F1675 will work.

SBS01G and SBS02G wont work.

wienne, 2020-04-06:
Using the latitude and longitude values of Berlin in your code, the display on the remote control is correct.

I use the ATGM336H-5N GPS chip made in China.
Information link of the chip:
http://www.icofchina.com/d/file/xiazai/2018-01-08/9c186bd617e127ca162479c480092300.pdf

The coordinate standard returned by GPS is NMEA01843,
N 3029.52785, the format is ddmm.mmmm, 30 is degrees, 29.52785 is minutes
E 11425.15706, the format is dddmm.mmmm, 114 is degrees, 25.15706 is minutes

Using this coordinate value, the position mark is very correct.。

[cid:image003.png@01D60C2C.B2117070]

ericlangel, 2020-04-06:
> The coordinate standard returned by GPS is NMEA01843,
N 3029.52785, the format is ddmm.mmmm, 30 is degrees, 29.52785 is minutes
E 11425.15706, the format is dddmm.mmmm, 114 is degrees, 25.15706 is minutes

Please try:

30295278 -> 4Digits after decimal point
114251570 -> 4 Digits after decimal Point

i think your Integer Value is just factor 10 to big.

wienne, 2020-04-06:
After updating the 4 sbus files, using GPS degrees and minutes, the display is normal.

  send_F1675 (GPS_SLOT, (uint16_t) GPS_SPEED, (int16_t) GPS_ALTITUDE, (int16_t) GPS_VARIO, (int8_t) GPS_LAD, (int32_t) GPS_LAM3, (int8_t) GPS_LOD, (int32_t) GPS_LOM3);

When testing outdoors, walking distance and speed are also normal.
It will be installed on airplanes and vehicles later for more tests.

Thank you very much!

BrushlessPower, 2020-04-07:
should be solved with V1.0

BrushlessPower, 2020-04-20:
Solved with V1.0 Release.

closed 2020-04-20 by BrushlessPower