BrushlessPower/SBUS2-Telemetry #20: Flow meter for fuel measurement

trunet, 2020-07-16:
Thanks for this nice library, I'm starting to play with it. I have a Futaba 10J and am planning to add a flow meter ([like this one](https://www.conrad.com/p/bt-bio-tech-fch-m-pp-lc-low-flow-flowmeter-corrosive-chemicals-npn-155374)) between the fuel tank and the engine.

Do you think this could work?
Which sensor do you think is best to use to push the telemetry data to my transmitter, voltage or current?

BrushlessPower, 2020-07-16:
Hello Trunet,

i never used a flow-sensor, but it should work without problems

You have to use a sensor which is available in your 10J (and supported by this library). 
You could just use a Tempsensor (Temp125) to display l/min -> but you cant change that it's displayed as °C
or you could use the Current Sensor: Voltage (V) for Tank Volume, Current(A) for actual flow and Capacity (mAh) for drained Fuel .... but the Unit's will be wrong.

http://www.vspeak-modell.de/en/flow-meter
they used your Sensor and Telemetry Temp Sensors for Fuel in % and flow in ml/min

trunet, 2020-07-16:
great... thank you very much. I will close as this was more like a question (not the best use for github issues).

closed 2020-07-16 by trunet