BrushlessPower/SBUS2-Telemetry #29: Frsky f-port support

stanislavdavid, 2020-12-06:
Hello, is it possible to make it compatible with frsky fport and fport2? :-)

BrushlessPower, 2020-12-07:
Hello,

i have no idea how F-Port(2) is working. I've looked at some specs, but i think F-Port2 is very different to SBUS2
And i have no TX/RX to do any tests.

If you have detailed Specs for F-Port2 which show that F-Port2 is similar to SBUS2 i can help you to develop prototype Code for that.

But actual i think this issue/feature request is over

stanislavdavid, 2020-12-07:
I found documentation about fport, if it helps you. I do not understand how to make it.
https://github.com/betaflight/betaflight/files/1491056/F.Port.protocol.betaFlight.V2.1.2017.11.21.pdf

BrushlessPower, 2020-12-07:
yes, thats the spec for F-Port (not F-Port2 i think)

for my understanding: F-Port just uses SBUS Data Format for their Control Frame. This can be received easily with Serial and decoded with all the available SBUS Libs

So you don't need a SBUS2 Library for that.

All the other F-Port Frames seems to be different to SBUS or SBUS2
The only thing is, that the upload Frame has to be send 3ms after download Frame. That's similar to SBUS2

It would be possible to rewrite my Library for F-Port2
But i can't do this because i have no Frsky system and no Time.

closed 2023-04-24 by ericlangel