BrushlessPower/SBUS2-Telemetry #30: [ESP32] ISR IRAM_ATTR Flash Issue

BrushlessPower, 2020-12-07:
There is a Bug for using ESP32 with this SBUS2 Library and write/read internal Flash

If ESP32 does a Flash read/write and a Interrupt occure (For Example receiving SBUS(2) or transmit Telemetry) the ESP32 crashes.
This is a known Bug, but i never thought about.

All SBUS(2) ISR (and it's called functions) have to run in IRAM

i will do some test and upload the fixed Bug.
Maybe it related to Issue #21 there where also mysterious Crash while using I2C

ericlangel, 2021-11-22:
i have to test every piece of the Library to make it run in IRAM

So i will do a complete new code.
1. UART Interrupt -> Running in IRAM
2. Timer Interrupt -> Running in IRAM

ericlangel, 2022-03-10:
i did some work on this Topic

i could get the Interrupts running in IRAM
but #33 needs some changes which would not work with IRAM

So i'am not sure which issue has more priority
I have to do some more work on that Topic

ericlangel, 2024-07-09:
on a local project i have changed the SBUS2 library without running in IRAM
every code line is executed in normal RAM/Flash and it is working

of course, if you are doing flash writes no SBUS2 Frames are processed.
so i will do the same here