BrushlessPower/SBUS2-Telemetry #33: Telemetry not working with 3 sensor

magnum3131, 2021-06-20:
Hi,

I use your library to make some sensor ( Temperature, GPS,...). if i plug only 2 sensors on the receiver with different slot everything is working perfectly.

But if i plug 3 sensors the telemetry on esp32 stop working, i have to stop one oh the three sensor to make it working again

Is there a limitation or something wich can explain these issue ?

Thanks
Julien

BrushlessPower, 2021-06-21:
Hi Julien,

interesting issue. Never thought about using multiple ESP32 (or Atmega) Sensors.

i have an idea whats causing the issue:
the Sensors (the third) could trigger the Start/End of a SBUS(2) Frame in the other Sensors. That would cause many problems.

If i'am right the same should happen with a original Futaba Sensor.
i have to check the code and try to get the same behavior.

Could yo tell me with Sensors you use on which slot and which ESP?
Example:
ESP32 No.1 -> temp sensor Slot2, Voltage Sensor Slot3
ESP32 No.2 -> GPS Slot 8
ESP32 No3. -> .....

Do the ESP32 anything? Are they crashing? Error Messages?

magnum3131, 2021-06-21:
Hi, 
Thanks a lot for your answer
ESP32 1 => Slot 1 rpm sensor
ESP32 2 => Slot 3 rpm sensor
ESP32 3 => Slot 5 rpm sensor

I made a loop wich increase the RPM, i can see when the telemetry stop working. When i plug the third sensor the telemetry stop working immediatly. there is no message or crach on ESP32.

I'll test this evening ig i have the same problem if i plug a futaba sensor

Thanks
Julien

magnum3131, 2021-06-21:
I did more test 
The configuration is the following
Powerbox Spark Switch pro => slot 1 to 5
Powerbox Mercury => slot 8 to 23
ESP32-1 RPM => slot 6
ESP32-2 RPM => slot 7

Case 1
Powerbox Spark Switch pro + Mercury + ESP32-1 => Working

Case 2
Powerbox Spark Switch pro + Mercury + ESP32-2 => Working

Case 3 
Powerbox Spark Switch pro + Mercury + ESP32-2 + ESP32-2 => Powerbox Sensor working but not the ESP32.

With other telemetry sensor i can only plug one ESP32

Thanks
Julien

BrushlessPower, 2021-06-22:
Dear Julien,

thank you for testing. I have 3 ESP32 and i will build up the same test an try to find the problem.

I think it could be a Problem with the 1k resistor between RX and TX of every ESP32 on the SBUS. But i can only check it with an oscilloscope.

Cold you try another test (before i finish my test build)?
ESP32-1 with 2,3,4,5 RPM Sensors on SLOT 1 to 6
ESP32-2 with just one RPM Sensor on SLOT 7

You said "the Telemetry stop working". So my understanding is, that all three RPM Sensors stop working? This would really point to a problem with the 1k resistor.

magnum3131, 2021-06-22:
Hi,
I'll test the new case this evening
I don't know if it could affect the SBUS2 Library but i have a 2.4 resistor between the TX/RX Pin

Thanks
Julien

BrushlessPower, 2021-06-22:
hi,
2k4 Resistor is ok.
I had problems with a R7003SB. This one only worked with 1k. Thats why i recommend 1k.

Maybe its better to use a diode.

magnum3131, 2021-06-22:
This is strange, the test with powerbox and esp32 has been done with a 7003sb on my plane 

I'll try with a 7008
Julien

magnum3131, 2021-06-22:
Hi,
I do your test 
ESP32-1 with 2,3,4,5 RPM Sensors on SLOT 1 to 6
ESP32-2 with just one RPM Sensor on SLOT 7

It's working with the 7008sb but not with the 7003sb, only one sensor work with the 7003 

You said "the Telemetry stop working". So my understanding is, that all three RPM Sensors stop working? This would really point to a problem with the 1k resistor. => Yes all three RPM stop working

Do you think we need to lower the value of the resistor ?

Thanks
Julien

BrushlessPower, 2021-06-23:
Ok

i really think it's the value of the Resistor. You could try to increase it when using multiple ESP32
Today i will have a look on the scope.

The perfect solution would be a one wire Serial with switching RX/TX on one Pin.

BrushlessPower, 2021-06-23:
So i got a View on the Scope and got back to the old Problem with 1k on R7003

1. The Resistor Value has to be small enough that the TX Pin can Pull High Signal on the SBUS (when transmitting Telemetry)
2. The Resistor Value should be as big as possible that the Servo Data is not pulled down to much (when receiving SBUS Frame and other telemetry)

for the R7008 4k7 and 1k was ok on my test
for the R7003 1k was ok on my test

But, when we connect multiple ESP32 on the same SBUS the Resistor Values are not valid any more

with three ESP32 (each with 1k) the Pull Down Resitance is 1k/3 = 333R -> thats much to low
and the pull up resistance i still 1k for each ESP32 to transmit his slot. But it has to work against the other 2 ESP32 with 500R

so at the End: The resistor Solution is not the Best
Did you have some diodes there?
You should add a diode between TX and the 1k. Anode to TX, Cathode to Resistor.

TX------>|----1k-----SBUS

RX-------------------SBUS

The Diode should have a low Forward Voltage Drop. I have tested it with a BAV21 and a high Drop (1V Voltage Drop) and it's working.
[image]

magnum3131, 2021-06-23:
Hi,

Thanks a lot for your expertise, for resuming i have to change my 2.4k resistor for a 1k and i have to add a diode. I need to place an order to test all this.

For the diode i found the 1N4148 which have a drop down of 0.7 or maybe a schottky diode with a lower drop down, I need to check a reference for a schottky.

Thanks a lot, I'll let you know when i'll have everything for testing

Other question what kind of oscilloscope do you use ?

Thanks
Julien

BrushlessPower, 2021-06-24:
The 1N4148 should be fine

Maybe you can also order a Schottkey with a high reverse Voltage (more than 10V). But this is something about try an error.

I use a PicoScope 2405A.
But the cheap PicoScope 2204A is also fine. But 2 Channel are not enough for my tasks.

Did you notice Issue #34 ?

magnum3131, 2021-06-24:
Hi,

I order all the stuff, i hope i'll receive next week. I had to find a solution because in my club we try to do ESP32 hardware which can work with Futaba (1k on TX) and Jeti (Jeti need a 2.4k resistor on the RX pin)

For the arduino core I am in 1.0.4, the 1.0.6 is available i hope the bug is not present in this version

Thanks a lot
Julien

BrushlessPower, 2021-06-25:
The solution with diode should also work with with 2k4.

i have the feeling that the 1.0.6 is more stable. I had a lot of issues with 1.0.5 and the are fixed with 1.0.6

p.s.
withe Jeti Telemetry you mean EXBUS?
Not EX?
until now, i cant find a good solution for ESP32 with EX Telemetry because the ESP32 cant handle 9 Data Bits

magnum3131, 2021-06-25:
Yes ex bus
I use this library 
https://github.com/Sepp62/JetiExBus
Julien

BrushlessPower, 2021-07-19:
any news here?

magnum3131, 2021-07-22:
Hi,Not yet, i'am in HallydayI'll let you KnowlesThanksJulien

ericlangel, 2021-07-23:
@magnum3131 

could you try the new Version of my library?
Some issues are solved (nothing real important)
and a new Feature is coded: reading Sensor Values of all SBUS Sensors on the Bus

and you have some ESP32 on the SBUS and a Powerbox Spark Switch pro and Powerbox Mercury

magnum3131, 2021-07-28:
Hi Éric,Of course i can, What kind of informations do you need ? Thanks Julien

ericlangel, 2021-07-28:
i have attached the new library and an Example sketch.

please change the SBUS2 Pins to your used SBUS2 Pins.
leave Pin 16 and 17 unconnected -> that is another topic

Now use my SBUS2 Library as usual.
You will get some prints for Sensor Slot 0 to 7. You can extend it to Slot31
The Output on console should show the value used  for the emulated RPM Sensor
And Slot0 (RX Voltage) and your other Sensors with their Slot Number

If that is working, we can go to the next Topic.

[SBUS2.zip](https://github.com/BrushlessPower/SBUS2-Telemetry/files/6891598/SBUS2.zip)

ericlangel, 2021-08-03:
@magnum3131 

i have a new Library Version which should work with 1k resistor and multiple Sensors (without diode or Voltage loss)
the TX Pin is just active (OUTPUT) when sending Telemetry Data. After that its disabled (INPUT/Floating).

It's not fully tested, but the first Test looks good

Please tell me if you want to test?

magnum3131, 2021-08-03:
Hi Eric,
Yes i want to test, if it's working i don't need to change my hardware i have already done
I'll be able to test it this weekend
Thanks a lot
Julien

ericlangel, 2021-08-03:
[SBUS2.zip](https://github.com/BrushlessPower/SBUS2-Telemetry/files/6924503/SBUS2.zip)

here the update
for explanation: the example sketch also sends sbus2 Frames on Serial2 -> so leave Pin 16 and 17 unconnected or use your sketch and just update the sources

magnum3131, 2021-08-03:
Thanks a lot,
I have to modify it because i use Pin 16&17 on my ESP32 
Julien

ericlangel, 2021-08-03:
ok, you could erase all Serial2 code in the example sketch

or use your old sketch and just update the library sources

ericlangel, 2021-08-04:
tonight i realized that the zip file i gave you wont work.

because Change TX to Float and back to TX is hardcoded to GPIO 26
i will change it and upload again

magnum3131, 2021-08-07:
OK thanksJulien

magnum3131, 2021-08-08:
Hi Eric,
I found the line with pinMatrixOutDetach and pinMatrixOutAttach, i adapted it to my pin.
I have the same problem with your library, it's only working with 2 ESP32 and some time with only one
The error are :
Error! -> rx_fifo_len = 128 Value = F, 0, 4, 60, 58, F8, 7, 96, BC, 1, 10, 80, 0, 4, 20, 0, 1, 8, 40, 0, 2, 10, 80, 0, 34, 0, 0, 0, 0, 0, 0, 4, 0, 4, 61, 1, FC, 3, 60, 1, 79, 3, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 0, 0, 0, 0, 0, A4, 3, 40, 3F, 0, 0, 0, 0, 6C, 3, 40, 3F, 0, 0, 0, 0, E8, 3, 0, 0, 0, 0, 0, 0, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 6C, 3, 40, 3F, 0, 0, 0, 0, E8, 3, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0,  [HEX]Error! -> rx_fifo_len = 128 Value = F, 0, 4, 60, 58, F8, 7, 96, BC, 1, 10, 80, 0, 4, 20, 0, 1, 8, 40, 0, 2, 10, 80, 0, 34, 0, 0, 0, 0, 0, 0, 4, 0, 4, 61, 1, FC, 3, 60, 1, 79, 3, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 0, 0, 0, 0, 0, A4, 3, 40, 3F, 0, 0, 0, 0, 6C, 3, 40, 3F, 0, 0, 0, 0, E8, 3, 0, 0, 0, 0, 0, 0, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 6C, 3, 40, 3F, 0, 0, 0, 0, E8, 3, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0,  [HEX]Error! -> rx_fifo_len = 128 Value = F, 0, 4, 60, 58, F8, 7, 96, BC, 1, 10, 80, 0, 4, 20, 0, 1, 8, 40, 0, 2, 10, 80, 0, 34, 0, 0, 0, 0, 0, 0, 4, 0, 4, 61, 1, FC, 3, 60, 1, 79, 3, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 0, 0, 0, 0, 0, A4, 3, 40, 3F, 0, 0, 0, 0, 6C, 3, 40, 3F, 0, 0, 0, 0, E8, 3, 0, 0, 0, 0, 0, 0, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 6C, 3, 40, 3F, 0, 0, 0, 0, E8, 3, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0,  [HEX]Error! -> rx_fifo_len = 128 Value = F, 0, 4, 60, 58, F8, 7, 96, BC, 1, 10, 80, 0, 4, 20, 0, 1, 8, 40, 0, 2, 10, 80, 0, 34, 0, 0, 0, 0, 0, 0, 4, 0, 4, 61, 1, FC, 3, 60, 1, 79, 3, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 0, 0, 0, 0, 0, A4, 3, 40, 3F, 0, 0, 0, 0, 6C, 3, 40, 3F, 0, 0, 0, 0, E8, 3, 0, 0, 0, 0, 0, 0, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 6C, 3, 40, 3F, 0, 0, 0, 0, E8, 3, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0,  [HEX]Error! -> rx_fifo_len = 128 Value = F, 0, 4, 60, 58, F8, 7, 96, BC, 1, 10, 80, 0, 4, 20, 0, 1, 8, 40, 0, 2, 10, 80, 0, 34, 0, 0, 0, 0, 0, 0, 4, 0, 4, 61, 1, FC, 3, 60, 1, 79, 3, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 0, 0, 0, 0, 0, A4, 3, 40, 3F, 0, 0, 0, 0, 6C, 3, 40, 3F, 0, 0, 0, 0, E8, 3, 0, 0, 0, 0, 0, 0, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 6C, 3, 40, 3F, 0, 0, 0, 0, E8, 3, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0,  [HEX]Error! -> rx_fifo_len = 128 Value = F, 0, 4, 60, 58, F8, 7, 96, BC, 1, 10, 80, 0, 4, 20, 0, 1, 8, 40, 0, 2, 10, 80, 0, 34, 0, 0, 0, 0, 0, 0, 4, 0, 4, 61, 1, FC, 3, 60, 1, 79, 3, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 0, 0, 0, 0, 0, A4, 3, 40, 3F, 0, 0, 0, 0, 6C, 3, 40, 3F, 0, 0, 0, 0, E8, 3, 0, 0, 0, 0, 0, 0, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 6C, 3, 40, 3F, 0, 0, 0, 0, E8, 3, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0,  [HEX]
ThanksJulien

BrushlessPower, 2021-08-09:
hi julien,

i forgot to upload the new version.
There are so many changes the last weeks.

I have no plan which version u are using

i upload as soon as possible

magnum3131, 2021-08-11:
Hi,Thanks a lot. I'll wait the new version for testing with multiple esp32Julien

ericlangel, 2021-08-17:
Here the actual Version
It's like a pre release, but i have not tested with multiple Sensors.
There are now some more Examples for the different use cases.
Atmega328P is not updated
[SBUS2.zip](https://github.com/BrushlessPower/SBUS2-Telemetry/files/6997560/SBUS2.zip)

magnum3131, 2021-08-17:
Hi Éric,I'll test it this evening Thanks Julien

ericlangel, 2021-08-17:
[SBUS2.zip](https://github.com/BrushlessPower/SBUS2-Telemetry/files/6997805/SBUS2.zip)

This one works just with 1k between RX and TX and both RX's connected
ESP32 No1 = Simulation Mode + Slot 7
ESP32 No2 = Normal Mode with Slot 1 to 19 (without Slot7)

All Slots are displayed
All Signal Levels are ok
removed Error Prints

magnum3131, 2021-08-18:
Hi Eric,
The result of the test :ESP32-1 : Port 1 RPM sensorESP32-2 : Port 2 RPM sensor
ESP32-3 : Port 3 RPM sensor

With a 2.4K between RX and TX (My esp32 were already soldered with a 2.4k)
It only works with 1 ESP32, the SBUS 2 stop working at the time i plug a second ESP32
I'll do more test this evening with a 1k resistor and also with a diode
ThanksJulien

ericlangel, 2021-08-18:
i will test it with a real futaba RX (not ESP32 Simulation Mode)

maybe something else was crashing. i have to make my sbus cable with a plug. so that i can dis/connect the ESP32 to see if there are faults

ericlangel, 2021-08-18:
hi julien,

i had the same issue. When connecting/disconnecting the ESP32 to the SBUS2 Port the USART Interrupt crashed.

i fixed that issue:
[SBUS2.zip](https://github.com/BrushlessPower/SBUS2-Telemetry/files/7005826/SBUS2.zip)

For Future Test: Enable the CoreDebugLevel for ESP32 to "Info". Then you will get Error Messages when something goes wrong on the SBUS Stuff.

i had a look on the Scope with ESP32, 1k and R7003
SBUSFrame (and Slot0) are at 3.3V
The RPM Sensors from ESP32-1 and ESP32-2 are at 2.0V

That's OK
But i would recommend 1k when using R7003.
With R7008 the 2k4 could be fine.

What i want to say.....in my DIY ESC with Futaba and Jeti Telemetry i have used 1k. And the Jeti Telemetry is fine with it

magnum3131, 2021-08-18:
Hi Eric,
It's better with the new version. it's working with 4 ESP32 (Slot 1,2,3, 4) on R7008SB with a 2.4K resistor.
Sometime it's locking few second and the following error is displayed (ESP32 in info mode)
I][SBUS_usart_ESP32.cpp:284] uart_intr_handle(): [SBUS2][Info] Receive Error[I][SBUS_usart_ESP32.cpp:284] uart_intr_handle(): [SBUS2][Info] Receive Error[I][SBUS_usart_ESP32.cpp:284] uart_intr_handle(): [SBUS2][Info] Receive Error
I'll try this evening with a R7003SB with in installed on my plane with other sensor (Powerbox Mercury & Spark switch Pro)

Thanks a lot
Julien

ericlangel, 2021-08-18:
Nice that it is working

this error should just occur when connecting/disconnecting an ESP32 to the SBUS.
it really should not occur in normal operation.

could you look how often it is displayed?
everytime the same ESP?
it could be also an issue of not enough signal voltage (maybe better to use 1k?)

magnum3131, 2021-08-19:
Hi Eric,
I did more test on my plane, the equipement is the following one :2 x R7003 (connected to powerbox Mercury, Telemetry connected to the 7003 Master)Port 1 to 5 => Powerbox Sparkswitxh ProPort 8 to 23 => Powerbox Mercury with GPSPort 24 => ESP32 with 2.4k resistor => Working wellPort 25 => ESP32 with 2.4k resistor => Working until there is change state on sparkswitch proThe errors are[I][SBUS_usart_ESP32.cpp:350] uart_intr_handle(): [SBUS2][WARNING] Invalid Slot ID[I][SBUS_usart_ESP32.cpp:284] uart_intr_handle(): [SBUS2][Info] Receive Error[I][SBUS_usart_ESP32.cpp:350] uart_intr_handle(): [SBUS2][WARNING] Invalid Slot ID[I][SBUS_usart_ESP32.cpp:284] uart_intr_handle(): [SBUS2][Info] Receive Error[I][SBUS_usart_ESP32.cpp:350] uart_intr_handle(): [SBUS2][WARNING] Invalid Slot ID[I][SBUS_usart_ESP32.cpp:284] uart_intr_handle(): [SBUS2][Info] Receive Error[I][SBUS_usart_ESP32.cpp:284] uart_intr_handle(): [SBUS2][Info] Receive Error[I][SBUS_usart_ESP32.cpp:284] uart_intr_handle(): [SBUS2][Info] Receive Error[I][SBUS_usart_ESP32.cpp:284] uart_intr_handle(): [SBUS2][Info] Receive Error[I][SBUS_usart_ESP32.cpp:284] uart_intr_handle(): [SBUS2][Info] Receive Error[I][SBUS_usart_ESP32.cpp:284] uart_intr_handle(): [SBUS2][Info] Receive Error
Port 26 => ESP32 with 2.4k resistor => same comportment of slot 25 with same error, i thing that 2.4k resistor is too mutch. i'll try with 1K
ThanksJulien

ericlangel, 2021-08-23:
Hi Julien,

the Errors seems to be Bytes that are not read correctly. I think that is an issue from the 2k4.
It would be really great if you could test it again with 1k.

I will add some more detailed Error Messages to the Log to make the debugging easier.
I will upload it to the git repo for everyone. It seems that it is stable enough for a release

magnum3131, 2021-08-25:
Hi Eric,
I change the resistor to a 1K, it's working with 4 ESP32. I have to check because sometimes the ESP32 lock on SBUS2_Setup and loop on reboot.
I'll try to debug it
Thanks Julien

magnum3131, 2021-08-28:
> If you write the EEPROM before SBUS_Setup() it is OK.
>
> if you want to write EEPROM in Runtime, you have to disable SBUS before write and enable after write EEPROM
>
> see #30
>
> i will try to test it here with 2 ESP's
> the problem with plugging is:
>
> maybe you plug in (or sbus is ready) in the middle of a Frame.
> This should causes an invalid frame and should be fine with the next Frame.
>
> But i'am actual not sure what the timer Intterupt do
> I think it is started right after sbus_stetup and maybe crashes....i have to test it.

Hi,
I clear my code in order to do all my read and write EEPROM before the SBUS2_SETUP
it's still crashing when the radio is on and i switch on the ESP32 connected to the receiver.
The error is the following one (occurs 1 to 9 times)Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)Core 1 register dump:PC      : 0x40144cc0  PS      : 0x00060034  A0      : 0x4008ebbc  A1      : 0x3ffc0580  A2      : 0x3ffc31a8  A3      : 0x3ffba264  A4      : 0x00000000  A5      : 0x40094b98  A6      : 0x00000000  A7      : 0x1300025c  A8      : 0x800812e0  A9      : 0x3ffc0560  A10     : 0x00000002  A11     : 0x00000101  A12     : 0x00000001  A13     : 0x00001000  A14     : 0x3ffc7e04  A15     : 0xffffefff  SAR     : 0x00000010  EXCCAUSE: 0x00000007  EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  Core 1 was running in ISR context:EPC1    : 0x4008f344  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40144cc0
ELF file SHA256: 0000000000000000
Backtrace: 0x40144cc0:0x3ffc0580 0x4008ebb9:0x3ffc05a0 0x4008f341:0x3ffba190 0x40085ae3:0x3ffba1b0 0x4009247a:0x3ffba1d0
PC: 0x40144cc0: uart_clear_intr_status at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/uart.c line 310
EXCVADDR: 0x00000000

Decoding stack results
0x40144cc0: uart_clear_intr_status at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/uart.c line 310
0x4008f341: spi_flash_op_block_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c line 82
0x40085ae3: ipc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c line 62
0x4009247a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
Thanks
Julien

ericlangel, 2021-08-30:
hi magnum,

first: we should move this to direct email. because i would like to see your testsketch and try it here

email: brushlesspower@protonmail.com

second: what do you mean with " i switch on the ESP32 connected to the receiver"?

is the ESP32 powered by receiver 5v?
is the receiver powered by ESP?
different power sources?
SBUS2 Cable already connected?
a lot of things.....

i tried it with connecting/Disconnecting SBUS cable -> no crash
i tried with plug in ESP32 to receiver (3pol servo cable)
i didn' try powering esp32 and receiver from different sources.....

Please write me an email with some more informations what you are doing and with your code. Then i can reproduce it here.