BrushlessPower/SBUS2-Telemetry #35: Frame loss

nightfury-11, 2021-06-29:
Will this library allow me to also get frame loss and holds etc on the radio? 

Second, do you have a wiring schematic for Arduino itself? 

Third, is the code provided for all the supported sensors? if so, the sensors that I do not want to use, can I remove that section of code?

BrushlessPower, 2021-06-30:
hello,

1. the library returs a Frame Error Rate ->SBUS2_get_FER()
this Value is a value from 0 to 100% and show the count of the last 100 received Frames (wireless transmission)
But this is not very linear. So if 20% of the Frames are lost, you are very near to loose your wireless transmission completely

2. The wiring is described in the Readme. For the Arduino it's more complicated, because you need a 2 channel inverter circuit, connected to RX and TX of your arduino.

3. the Code is for all the supported Sensors. The Example.ino just show some of them. in the SBUS2.h you will find all Sensors(functions).
In your sketch just remove or add the Sensors you want or don't want. You can also paste your code here if you have Problems.

4. I'am not sure if i can translate it into words. But i try my best:
All Arduino's (with Atmega328P) can't invert the Serial Data. The Problem is that the SBUS(2) Signal is inverted. So you can use the ESP32 which is able to invert the Data or you have to do it with an external circuit.
For Receive SBUS(2) the SBUS signal has to be inverted with a Transistor (Mosfet or NPN). just connect the Gate to SBUS and the Drain/Collector to your RX Pin.
Same for transmitting SBUS2. Your TX Pin connect to the Gate of the Second Transistor. And the Collector with a 1k Resitor to SBUS(2). For sure you need some resistor for Pull Up/down and Gate resistor. But this Values are more variable.

nightfury-11, 2021-06-30:
Understood point number 2, 3, 4. the schematic is much clearer now. 

for point number 1, is there any way that all the frames instead of the last 100 be surveyed? and the total number of lost frames be sent to the radio?

BrushlessPower, 2021-06-30:
STM32 is not supported. If you have some experience with STM32 you can try to implement the code. I will help you and do the tests here.
STM32 is able to invert the signal. It also would be a great thing to support stm32 because its used very often in the arduino world.

0.5A transistor is fine. Even a 10mA should work ;)

Frame error count can be done

1. you could change the library sources. In an older Version SBUS2_get_status() returned the value of all  lost frames. actual it returns if the last frame was lost.

2. you could check SBUS(2)_ready() -> if yes a new SBUS Frame is received (via Serial link from Receiver). And then ckeck SBUS2_get_status() to see if the Frame was lost or not (Wireless from your Transmitter)
But you have to check this very fast. Your loop time should be under 10ms

nightfury-11, 2021-06-30:
> STM32 is not supported. If you have some experience with STM32 you can try to implement the code. I will help you and do the tests here.
> STM32 is able to invert the signal. It also would be a great [...]

nightfury-11, 2021-06-30:
Ok, understood. Using the older library would be easier in my opinion, however will that affect the remaining sensors if at all they are being used? 

second, while using arduino, can I use the software serial library to reverse the signal or will I have to do it with the transistor and resistors?

BrushlessPower, 2021-06-30:
i just checked the code. i think i was wrong

lost SBUS Frame (wired transmission) is counted up (and was never change)
lost SBUS Frame (wireless transmission) is just the last Frame (and was never changed) (true or false)
so you have to change the code in the lib sources. But it's really easy. I can send you the lines of code
You should really use the newest Version

you cant use Software Serial because you need 100.000 Baud. And Softwareserial is to slow for it

nightfury-11, 2021-06-30:
Can I use the newest version yet at the same time be able to count lost frames?? Please do send me the code, I don't want to mess up the library. The code will go into the "library.properties" file correct? 

the newest software serial update supports 115,200 baud, is that too little?

BrushlessPower, 2021-06-30:
you have to change SBUS_usart_328P.cpp

line 420 to 426

  if (sbusData[23] & 0x04){
    FER_buf[FER_count] = 1;
    FER_count ++;
    if(FER_count > 99){
      FER_count = 0;
    }
  }

here you have to add your variable (for example errorcounter) and let it increase

  if (sbusData[23] & 0x04){
    FER_buf[FER_count] = 1;
    FER_count ++;
   errorcounter++;
    if(FER_count > 99){
      FER_count = 0;
    }
  }

of course you have to define the variable and write a get() function for it

you can use a local copy of the library.

forget softserial.
even it can **receive!** at 115200 Baud (i don't believe that) you will have other problems. Because you will need to get an interrupt when a Byte is received.
you can try it. If it is working it would be a great feature.

nightfury-11, 2021-06-30:
Ok, I will change the requisite lines. 

I'll also try using software serial to see if it can work, and will circle back here.

nightfury-11, 2021-07-02:
@BrushlessPower ?? any update on the instructions?

BrushlessPower, 2021-07-03:
i'am not sure hiw i can help you?

it seems that you have basic coding skills. So you should be able to get it working with the instruction i gave you.
If you have problems, you can paste the code or send a Repository with your code.

nightfury-11, 2021-07-06:
Yep, working on it right now. Should be done in the next few days.

BrushlessPower, 2021-07-19:
any news here?

nightfury-11, 2021-07-19:
yep, have coded it and the board is ready and installed. i only have to
test it at the field.

closed 2021-08-05 by ericlangel