BrushlessPower/SBUS2-Telemetry #36: Reading SBUS2 sensor data to serial monitor

i3dm, 2021-07-18:
Hello,

is it possible to use this lib to read SBUS2 sensor data to Serial monitor / PC?
i have an application where i like to take existing sensors data (sbus2) and display it on a PC.
thanks.

BrushlessPower, 2021-07-19:
Hello,
the library doesn't have functions for sniffing SBUS2 Sensor Data.
I'am not sure if stand alone Sensors will send any SSensor Data without SBUS Frame?!

But it should be possible if you are using a Futaba SBUS2 RX
You have two options:
1. Using normal Serial.read() to read all Bytes in loop() and decode them with the help of this library.
That would be the fastest to get the Sensor Data. But it would use  a lot of CPU time (no Problem with ESP32 and 2 Cores)

2. you could change my library.
- disable Transmit and transmit Timer
- change USART RX Interrupt to receive (and decode) Sensor Data
- implement some "sensorGet"() functions
That would be the smartest way and i could integrate it to the library.

If you want to use the Sensors stand alone, it will be more complicated. Because you have to Send SBUS Frame and Receive the Sensor Data.

If you have any questions, please contact me here.

i3dm, 2021-07-19:
what i would like the board to do is to poll the SBUS2, read data from attached sensor, and then print it to serial monitor.
Option #2 sounds like what i need but it is out of my reach to perform the necessary modifications to the lib - would you be willing to assist in that?
thanks.

BrushlessPower, 2021-07-20:
So you want to use a Futaba Sensor without Futaba RX.

The problem is, that this library is not made for sending SBUS(2) Frames. It's possible but there will be many changes in API.
The changes for reading other Sensors would be easy i think. That could be a nice Feature for this library.

But i have no spare time to do all these changes and test.

If you would use a Futaba RX you will have another Problem -> Without a Futaba TX the RX wont transmit any SBUS(2) Frames.
So....thats no Solution.
Mayby you could use the Bolderflight SBUS Library to transmit SBUS(2) Frames and just use another Serial for receiving?

The main Question for me: Why you want to use Futaba Sensors?

i3dm, 2021-07-20:
I occasionally have sensors that need to be tested on the ground, for now i just do it with a futaba Tx and Rx but i wanted to find a stand alone solution for testing them without an actual radio.

BrushlessPower, 2021-07-20:
ok, thats a good reason.

i could try to implement a function to get the raw Bytes on the Telemetry Slots

Slot = SBUS2_get_slot_data( 2 ) // get Second Slot Data

But you would have to calculate the Sensor Value from the Raw Slot bytes (sometimes from multiple Slots)
That means, that i have to make get functions for every Sensor supported in this Library
That is a lot of work.

i3dm, 2021-07-20:
calculating should not be an issue, as on ground testing i can control the sensors physical input :)
if you could add that, it would be really great!

BrushlessPower, 2021-07-20:
The Sensor Values are separated in Slots.
Every Slot has 2 Bytes of Data

A Temperature Sensor needs 2 Slots
A GPS Sensor needs 8 Slots
So you need to know which Sensor starts on which Slot and how many Slots are used for that type of Sensor

My Library calculate the Slot Bytes for every supported Sensor (every Sensor is calculated different)

For your use Case you need to reverse all my calculations. 
You can see the calculation in SBUS2.cpp

i3dm, 2021-07-21:
thats fine, i can calculate myself.
for now i only need the function:

slot_val = GetSlotData(SlotNum);

BrushlessPower, 2021-07-21:
ok, i will try to do it this week and will upload it

just to be clear: you will use ESP32?

i3dm, 2021-07-21:
thats very nice of you. thank you.
i can also use an aduino with an inverter but i should have some ESP boards at home so i think i will go for those. just need to figure out how to run them with Arduino IDE (first time using ESP32).

BTW will you also be able to include an example in the lib to read a sensor value?

BrushlessPower, 2021-07-21:
ESP32 would be much easier for me, because i have the Serial for Debugging (and you will need it too)

Arduino with ESP32 is very easy. You just have to install ESP32 Hardware Library (Version 1.0.6)

i think i will upload my test Sketch directly to Github Master without Release

i3dm, 2021-07-21:
sounds good. let me know when i can test it.
will you also return a flag is sensor data exists on a certain slot? thats useful too

BrushlessPower, 2021-07-21:
so i was able to change the library to save the Raw Telemetry Data to a buffer

actual i'am just print that buffer on Serial
here is the output (in Hex) of that buffer:

0 0 0 3 C4 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 23 4 D2 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 83 64 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 A3 3A 98 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 C3 49 29 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

Slot ID's
0x03 -> RX Voltage on SLOT 0
0x23 -> Current Sensor on SLOT 4 (starting with Slot3)
0x83 -> RPM Sensor on SLOT1
0xA3 -> Current Sensor on SLOT 5 (starting with Slot3)
0xC3 -> Current Sensor on SLOT 3 (starting with Slot3)

now i will implement the get function

BrushlessPower, 2021-07-21:
[SBUS2.zip](https://github.com/BrushlessPower/SBUS2-Telemetry/files/6854793/SBUS2.zip)

So here is a very first Solution.
The Example sends RPM on Slot1 with 600RPM and Read it back

you should be able to reduce that sketch down to your needs and read your Slot Data

i3dm, 2021-07-21:
i will test it on ESP32 and report back.
what is the pinout for SBUS2 on the ESP32?

BrushlessPower, 2021-07-21:
you can use every GPIO pin for RX and TX 

in the Example i have used 25 and 26. And GPIO 16 is used for Oscilloscope to see the USART Interrupt

you have to use a Futaba RX with SBUS2 and an active Radio
SBUS(2) Wire to RX (IO25) and 1k Resistor from TX (IO26) to SBUS(2) Wire

email: brushlesspower(at)protonmail.com

i3dm, 2021-07-21:
the purpose of the device i need is to be able to test the sensors without a working Futaba Tx (i dont mind the Rx as its small).
will this be possible somehow?

also sent you an email.
thanks.

BrushlessPower, 2021-07-21:
you cant run it without active TX.
Futaba RX (R7003 and R7008) doesn't send SBUS Frames until the TX is active.
When you switch off the TX, the SBUS Frame is still send, but with active Failsafe Flag.

So if you power your RX 24/7 you don't need a TX (just for the first startup)

i3dm, 2021-07-21:
the solution im trying to achieve is that the ESP32 will poll the SBUS2 sensors just like the Rx does, and my sensor will answer. then the data read back on ESP32 will be printed to serial monitor.

BrushlessPower, 2021-07-21:
what you could do
on Serial1 -> Debugging
on Serial2 -> My SBUS2 Library
on Serial3 ->Bolderflight SBUS Library for sending SBUS Frames (no idea if it will will work or if they can send SBUS2 Frames)

But it should be possible

Also you could just send SBUS2 Frames with Serial.write(sbus2[], Framelength); at 100.000Baud
the Servovalues doesnt have to make sense

i3dm, 2021-07-21:
> Also you could just send SBUS2 Frames with Serial.write(sbus2[], Framelength); at 100.000Baud
> the Servovalues doesnt have to make sense

does your lib support this?

BrushlessPower, 2021-07-21:
> does your lib support this?

No.
Because every Transmit (Telemetry Data) is controlled by Timer Interrupt.
It could be possible to change it.....but i'am not sure how to do it.
Futaba SBUS2 has very hard Timings. No idea what happens when Frames are not send correctly

So an easy way would be another Serial.

i3dm, 2021-07-21:
any chance to get an example from you that sends sbus2 frames to poll the sensors via serial print?

that way i can even use 2 ESP32. one for polling the sensors and one for sniffing the data bqck to the serial monitor.

thanks.
Lior.

BrushlessPower, 2021-07-21:
i have no example

SBUS is documented. You just have to search for it
Boldeflights SBUS Library is doing exactly what you want and gives many informations.

its an easy thing

if you want me to build a plug and play system i can make you an offer.

emilverwoerd, 2022-01-05:
Was reading this post and also want something like this. Want to add some additional sensors to this library. Like the hobbywing protocol so want to sniff it with an Y-cable. Between the original sensor and receiver

BrushlessPower, 2022-01-06:
hi @emilverwoerd,

with the master branch it is already possible (with ESP32). See this Example: https://github.com/BrushlessPower/SBUS2-Telemetry/blob/master/examples/SBUS2_Read_Telemetry/SBUS2_Read_Telemetry.ino

If you are sniffing Sensors please post your findings here. I can help you with decoding or add the Sensors to the Library

ericlangel, 2022-01-21:
@emilverwoerd 
any news?

emilverwoerd, 2022-01-21:
Hi @ericlangel  the code to sniff the package works great. I managed to send telemetry for the following sensors:
- Scorpion ESC
- Kontronik ESC
- (note that these are actually the same)

These sensors however are only on the more advanced transmitters. Currently I'm waiting for the hobbywing sensor which is in te latest firmware and than I can add that to. 

Just finished this week since I hadn't a spare ESP32 laying around

emilverwoerd, 2022-01-21:
I will do it later today

billis-xania, 2023-05-19:
Hello, first of all thank you for this library! I use the example for simulation in order to test some Futaba sensors without Futaba receiver, i am using it with ESP32, it works great.
I tried to make two simulations in the same time in order to check with one ESP32  two sensors.
I understand that i have to declare different serial port, in your example(SBUS2_Simulation(25,26)) i think you are using the UART1, so i tried to use UART0 or UART2 with the following code:
SBUS2_Simulation(25,26,2); 
or
SBUS2_Simulation(25,26,0);
I study your library (SBUS2.cpp) and i saw that this was wrong, so i tried the following that in my opinion it is better.
This worked: SBUS2_Setup(32, 33, 1);
But that didn't worked: SBUS2_Setup(32, 33, 2); or SBUS2_Setup(32, 33, 0);

Is there a way that i can make two simulations in the same ESP32, at the same time?
Thank's for your time!

ericlangel, 2023-05-22:
@billis-xania 
I'am not sure what you are trying. But you can't run 2 SBUS's on one ESP32.
The ESP32 could handle it, but the library is written in C. For this task it should be written in C++ as Object-oriented functions.

But i never saw a use case for that.
Maybe you could explain you use case?

billis-xania, 2023-05-24:
Thank you for your fast reply, I want to check multiple sensors in the same time.
I understand that you said about your library, finally i found another solution to my problem.
I binded them with one receiver, i change the beggining slots from Futana so now they work with the same Sbus Simulation frame.
I read all the slots from esp32 so i am ok with that.
Thank's again!

ericlangel, 2023-05-24:
Nice that it is working for you