BrushlessPower/SBUS2-Telemetry #40: [ESP32][BUG] Not working with ESP32 Arduino V2.0.0

BrushlessPower, 2021-09-27:
Code doesnt compile with ESP32 Arduino Core V2.0.0

BrushlessPower, 2022-03-10:
Yes,

i have a local version running with the actual ESP32 core V2.x.x

The are just some Variable names that changed from V1.X.X to V2.X.X.
These Names have to be changed.
I'll try to work on the new version next week. There are a couple of changes and Bug fixes to do.

i just looked at the Code: maybe it just one line:

in SBUS_usart_ESP32.cpp
in void SBUS2_uart_setup()

change

ESP_ERROR_CHECK(uart_set_line_inverse(uart_num, UART_INVERSE_RXD|UART_INVERSE_TXD)); 

to

#if(ESP_ARDUINO_VERSION_MAJOR == 2)
  ESP_ERROR_CHECK(uart_set_line_inverse(UART_NUM_2, UART_SIGNAL_TXD_INV | UART_SIGNAL_RXD_INV));         // V2.0.0
#else
  ESP_ERROR_CHECK(uart_set_line_inverse(UART_NUM_2, UART_INVERSE_RXD | UART_INVERSE_TXD));                 // V1.6.0
#endif

BrushlessPower, 2022-03-10:
It need some more lines

i will upload my local changes for testing in the next minutes

ericlangel, 2022-03-10:
it should work with the actual master branch

after testing i can make a release and make it available in library manager

emilverwoerd, 2022-03-10:
Thx for the quick fixes. I tried to compile it and that works. However with boad ESP32**C3** selected it doesn't work its missing stuff

ericlangel, 2022-03-10:
Oh dear...

never tried. I opened issue #44  for that