BrushlessPower/SBUS2-Telemetry #43: [Feature] Add JetCat V10 Sensor

ericlangel, 2022-03-10:
there is some decoding done

i cant check it with T14SG

But i will try to add the Sensor with the known Data

ericlangel, 2022-03-10:
@i3dm 
can you check the actual master branch with the jetcat Sensor?

i3dm, 2022-03-10:
> @i3dm can you check the actual master branch with the jetcat Sensor?

sorry my friend but i have no more Futaba radio.
this is why i needed the SBUS2 decoder to test my sensors, i am working with your decoder and it works well.
maybe connect your sensor to your decoder will work for testing?

ericlangel, 2022-03-10:
ah, ok.

no Problem. We will see if someone else will check the JetCat Sensor
I would like to see if it work on the real system.

kevinuav, 2024-08-05:
HI, Ericlangel. I have already successfully sent the RPM,EGT to jetcat's slot. But the status. I have tried to send the test status code. It doesn't respond to it. Why? Is there something preprocessing I miss before the status code sent into the "SBUS2_transmit_telemetry_data()"
Here is the status: 
[image]
This is the RPM,EGT:
[image]