BrushlessPower/SBUS2-Telemetry #44: [Feature] Support ESP32-S2 & ESP32-C3 & ESP32-S3

ericlangel, 2022-03-10:
actual these types cant compile because of missing stuff

Nellophonic, 2022-03-12:
Examples don't actually compile for ESP-32:

In function 'void setup()':
SBUS:16:20: error: too many arguments to function 'void SBUS2_Setup()'
   
SBUS2_Setup(25,26);     // For ESP32 set RX and TX Pin Number <<--- this doesn't work

emilverwoerd, 2022-03-15:
Tried to modify some code to get it to work with ESP32 C3 but didn't work out. There are some changes with the UART and Timers which makes it difficult. It also has less timers and uarts than the original ESP32

ericlangel, 2022-03-16:
yes,
i also had a look.
Some missing Defines are stored in different Headers. That was a quick fix.

But, the UART and Timers are coded on a very Low System Level. And these functions and variables are completly different to C3, S2 and S3.

1. Problem
there are not so much examples out there for Low Level UART and TIMER for C3,S2,S3

2. Problem
ESP32 Arduino is based on ESP-IDF
i think V2.0.x is Based on ESP-IDF V3.x
the actual IDF Version is 4.x
So my fear is, that the next Arduino Core Version will move to IDF4 and change the ESP32 Api to the same as C3,S2,S3
but it's just guessing.

Before i start to work on C3,S2,S3 i want to check what's the right Api to use
For Testing i have a ESP32-S3
And if i understadn you correctly you have a ESP32-C3

I'm happy about any help.
I would start to ask in the ESP Arduino Gitter Channel: https://gitter.im/espressif/arduino-esp32

ericlangel, 2022-03-16:
i managed to solve all UART Problems

just the Timer is left

ericlangel, 2022-03-16:
i also managed to get it compiled for S2
but not for C3

the Reason is pretty Simple: C3 does not have TIMER_1, just TIMER_0

So the we have to Check if we can use:

TIMER_0 on TIMER_GROUP_0
Or 
TIMER_0 on TIMER_GROUP_1

and maybe we can switch it for all ESP32 versions.

i attached the sources for you for testing
for C3 you have to Change TIMER_1 to TIMER_0
and maybe TIMER_GROUP_0 to TIMER_GROUP_1

but i have no idea if it is working. It's also not tested on ESP32

[src.zip](https://github.com/BrushlessPower/SBUS2-Telemetry/files/8261427/src.zip)

emilverwoerd, 2022-03-16:
I have solved the timer issues in my code tried to fix the UART issues also although it compiled it didn't work let me check you're code to

ericlangel, 2022-03-16:
did you have the exeption decoder installed?

From the output: the watchdog was triggered.
Not sure why.
does it happen wirh sbus cable attached?
try without sbus cable

i'll try to test it on the esp32 and S3

emilverwoerd, 2022-03-16:
I never looked into the exception decoder will try that. I have the same result with and without sbus cable attached.

emilverwoerd, 2022-03-16:
see still 1 issue when using the last slot of a group its not send so slot 15 isn't visible

ericlangel, 2022-03-17:
Nice to hear that it is working on the C3
did you have a normal esp32?

i have to check if the issue also exist in the other versions. could be a logical issue.

if you have a oscilloscope or logic analyser, you could toggle a pin in the timer ISR to check that the timer ISR is called 8 times after sbus frame

emilverwoerd, 2022-03-17:
Currently I don't have a osciliscope so thats currently out of the capabilities however maybe I can do something with Serial.println

Did the following code:
- Temp was visible on slot 14
- Temp was **not** visible on slot 15
- Voltage in slot 16 was changing values when delay was set at 100 sometimes temp value, sometimes voltage value

```send_F1713(14, 55);
        send_F1713(15, 55);
        send_scorpion(
            16, 
            telemetryData.voltage,
            telemetryData.consumption,
            (telemetryData.rpm / gearRatio),
            telemetryData.current,
            telemetryData.temperature,
            0,
            0,
            telemetryData.pwm
        );```

ericlangel, 2022-03-17:
Serial.print in an ISR is a no go.
I think it would crash the ESP.

i will check it with scope. Maybe i have to get a C3.
but i will need some days

emilverwoerd, 2022-03-18:
Hi thanks already that you're going to check it really happy with the support for making this work on a ESP32-C3.
I use the following board

https://shop.m5stack.com/products/m5stamp-c3u-mate-with-pin-headers

emilverwoerd, 2022-03-20:
Did a small test and when changing the following it works
Changed the slot time to 650ms

```
#define SLOT_TIME 650 // 660 ms
```

ericlangel, 2022-03-20:
Oh, thats great.
So its just a Simple Timing Issue.

Could you try with 640ms and 630ms to get the working range.

i'll try it with 650ms and 640ms on the normal esp

emilverwoerd, 2022-03-20:
the range to get it to work on my esp32-C3 was between 20ms and 650ms.  In this range it seemed to work correctly

ericlangel, 2022-03-21:
oh just to be sure

20ms is a typing error and you mean 620ms?

or really 20ms? This would be very interesting.

emilverwoerd, 2022-03-21:
Yeah you would think indeed that it is a typing error but no it really is 20ms so the bandwidth is extremely large have to note only tested it with a R7006SB have to check with other receivers

ericlangel, 2022-03-21:
SBUS2 is defined with the following Timings:

After SBUS Frame-> wait 2000ms
and then send the SLots every 660ms

so, if you go down to 20ms i would think that the Slots are send directly one after the other.

So maybe the SBUS2 protocol tolerate that
or there is "something" special in C3 Timing functions

ericlangel, 2025-10-06:
we are currently working on an updated version for better support of newer chips and Core Library

i think the S3 and S2 should work with the main branch.