BrushlessPower/SBUS2-Telemetry #47: How do you login your Arduino as a sbus2 device to your remote controller?

kevinuav, 2022-03-21:
Hi,
    Thank you for your work on the sbus2. But I don't see any code for logining ro the remote controller. Since every sbus2 device should be login to the remote controller by plugging the sensor into the remote controller first. Or it can't be display on the sensor's interface screen. How do you overcome for the problem?

ericlangel, 2022-03-21:
hi @kevinuav 
me and @BrushlessPower have no acces to the "login protocol". And it is not necessary.

normaly login:
the Sensor needs a Slot Nummer to send his telemetry. So remote Control is looking for a free Slot and set the Slot Number in the Sensor.

Manual Login:
You can set your Sensors manually in your Remote Control. You just have to tell your Code which Slot Numbers are used for which Sensor.

if you have problems, please post your Remote Control type.
I can help you with T14SG and i think its similar to other models.

kevinuav, 2022-03-22:
Hi,my remote controller is T18SZ. the receiver is R3001SB. I wanna get all my remote controller's control value and send back my telemetry data with just connecting 1 sbus2. Is sbus2's timing sequence equals sbus1 + the telemetry slots?

kevinuav, 2022-03-22:
Hi,my remote controller is T18SZ. the receiver is R3001SB. I wanna get all my remote controller's control value and send back my telemetry data with just connecting a sbus2. Is sbus2's timing sequence equals sbus1 + the telemetry slots?
So that I can read the control value as a sbus1 protocol. Then send back the telemetry data with the slots?

BrushlessPower, 2022-03-22:
@kevinuav 

please look at the Examples. The "Futaba_Telemetry" Example does exactly what you want

you can get your Channel Values with:
SBUS2_get_servo_data( 2 ); -> Example for Channel 2

Were you able to set your Sensors manually in your T18SZ?

kevinuav, 2022-03-23:
Hi,Thank you for your help.
    I inspect the sbus2 signal with the logic analyser. And I noticed a odd place. The every frame which ends with 0x04 comes with 3bytes. what are they?
    And BTW. Could I send my message directly after the 0x04(slot0-7),0x014(slot8-15).......... without any other operation?
[image]
[image]

ericlangel, 2022-03-23:
0x04 is Frame No.1
After Frame No. there comes SLOT 0- 7

0x03, 0xC4 and 0x00 -> is Slot 0 = Receiver Voltage

No, you cant send your SLOTS directly after the Frames. There are some strict Timings needed.

What are you trying to do?
Do you want to port the library to another MCU?

kevinuav, 2022-03-23:
Yeah I wanna transplant the code to STM32F7.

kevinuav, 2022-03-24:
Got it!
    I wanna port it to my STM32F7. But I still have a question about sbus2. Why the receiver's voltage data comes out from the receiver since the telemetry is sending back the data from sensor to RC? The direction of the data stream should be sensor->receiver->RC. However the data stream I got from logic analyses is receiver->sensor. Dose it work like this: receiver->sensor->receiver->RC? This confuses me.

ericlangel, 2022-03-24:
i think you have to ask Futaba, why they have the RX Voltage available on the SBUS2.

You can Send Slot 1 320ms after Slot 0 (RX Voltage)
Your Timer have to do it.
Let the Timer work also on Slot 0.

1. you can check the timing (toggle a pin on each timer interrupt)
2. you will need it for Slot 8, 16 and 24.

just dont set any Sensor on Slot 0.
This Library will do the rest for you

kevinuav, 2022-03-24:
OK. thank you for your help.

ericlangel, 2022-03-24:
please share your code when you are finished, or come back when you have problems

and we should continue in #32

closed 2022-03-24 by ericlangel

kevinuav, 2022-03-30:
Hi,thanks for your help.

kevinuav, 2022-03-30:
Hi,
    Thanks for you help. I can't send back the data for the average slot. But the jetcat's and the GPS's. Why? Aren't the GPS's and the jetcat's slot the same as the average?

kevinuav, 2022-03-30:
Hi,
Thanks for you help. I can send back the data for the average slot. But the jetcat's and the GPS's. Why? Aren't the GPS's and the jetcat's slot the same as the average?

ericlangel, 2022-03-30:
@kevinuav 
what is the average Slot?
i dont understand what you are saying.

p.s. could you get it working with STM32?

ericlangel, 2022-03-30:
why you want to receive the jetcat and gps sensor?
you should write them?

So if i understand you correctly, you are receiving the SBUS Frames with USART Interrupt?
do you receive every Frame?
Did you set up the timer to fire on every Slot? for the first stuff you dont need to Send telemetry Data. Just toggle a LED in the Timer interrupt to see an the Scope/logic analyser that the timing is correct.