BrushlessPower/SBUS2-Telemetry #50: It doesn't work with R7108SB

kevinuav, 2023-04-07:
I have successfully make it work on receiver R3001SB. However it can't work with R7108. The slot 8-slot 31 just can't work. But the slot0-slot7. Why?
I asked chatGPT what's the scheme of the sbus2. It says the sbu2 telemetry frame should have a sync pulse at the beginning of it. Why I can't see any code of the sync pulse in your code?

[image]

ericlangel, 2023-04-17:
Interesting answer from ChatGPT :)

The Answer is: No there is no Sync Pulse. SBUS2 is based on hard Timings

you could play with:
#define SLOT_TIME 660 // 660 ms
#define WAIT_TIME 2000 // 2000 ms

in SBUS_usart_ESP32.cpp

try in Range 1900 to 2100 for wait time
and/or 630 to 690 in Slot Time

i would start with the wait time

kevinuav, 2023-04-20:
Yeah. the 7108 is even more time strictly than the 3001. That's why the program can work on 3001 but the 7108. But how can the devices manufactured by the same company using the very same protocol, which's working condition is not the same. I can only make one firmware for the 3001 ,and another firmware for 7108. Oh no!
    BTW, ChatGPT's answer is not always believable. He will give you the wrong answer when he doesn't know it, rather than says "I don't know."

ericlangel, 2023-04-20:
Could you give me the Slot and Wait Time for the 7108?

All Timings are Reverse engineered. So maybe they are not perfect.
Or maybe Futabas Time calculation is based on other triggers?

If you send me the working timings for the 7108 i could check if i find any hint

kevinuav, 2023-04-20:
BTW, I can't imagine there are so many manufacturers producing sbus2 sensor in the world without the official technical document. Why can't we get a copy of sbus2's official document?

ericlangel, 2023-04-20:
i'll try to do some calculations.

my personal problem: i have zero original SBSU2 Sensors. So all timings are based on Internet findings and "Try and Error"

All manufactures of SBUS2 Sensors have the documents from Futaba. But they need to sign an NDA.

So....i could sign the NDA and get the Timings, but i wont be allowed to talk about or publish it on an open source project

kevinuav, 2023-04-20:
I know it is hard for a non-profit personal using purpose to get the document. I just fell strange about that. : )

kevinuav, 2023-04-20:
You said:"So....i could sign the NDA and get the Timings, but i wont be allowed to talk about or publish it on an open source project"  If you signed the NDA and develop the program. You can't publish its document only or even the program? Can we have any chance to avoid the regulation of the NDA?

ericlangel, 2023-04-20:
If you get a Working Setup for the 7108, please let me know the values.

i never tried to get the documents
and i wont try it in the future, because i dont want to have any trouble with NDA's and other stuff.

you could talk to Futaba yourself.