BrushlessPower/SBUS2-Telemetry #52: Send RPM value to receiver

billis-xania, 2023-10-03:
Hi, first of all big thanks to this amazing library! I want to make my own rpm sensor for my plane, i can send values to the futaba receiver from my esp32 but i have problem with rpm, for example i use "send_RPM(RPM_SLOT,(uint16_t) 500);" and in futaba i have the number 62 in this slot(of course i have set the slots and the sensors from futaba settings). I try with voltage and it works, i know that it must be something in conversion in the code or in futaba. Do you know what is wrong?
Thank you for your time.!

ericlangel, 2023-10-04:
Hello,
the RPM Sensor is one of the easy going sensors. There is not so much conversation.

So at first lets check your hardware:
Which Futabe TX?
Which Futaba RX? -> some RX are known for "more trouble"
Which Sensor is selected? -> there are several rpm sensors possible
Which ESP32? -> normal esp32, C2, C3, S2, S3?
Which ESP32 Arduino Core Version?

could you try 0 and 1 for the rpm value and tell me the displayed value?

billis-xania, 2023-10-04:
So i use: 	Futaba T16SZ(FASSTest 18ch)
		Futaba R7008SB
		I try with all of the RPM sensors in your code(send_RPM, send_SBS01RB, send_SBS01RM, send_SBS01RO, 
                send_SBS01R), in the Futaba Tx menu: "sensor" -> it has only one rpm sensor "rpm sensor"
		
		ESP32 DOIT DEVKIT V1 - esptool.py v4.2.1
		
		Arduino IDE 1.8.19
		In "board manager" 	-> "ESP32 BY ESPRESSIF SYSTEMS 2.0.5"
		In "board" 		        -> "ESP32 DOIT DEVKIT V1"

		For example :
		send_SBS01RB(RPM_SLOT,(uint32_t)0);		        ->		Futaba Tx(rpm sensor): 0
		send_SBS01RB(RPM_SLOT,(uint32_t)1);		        ->		Futaba Tx(rpm sensor): 0
		send_SBS01RB(RPM_SLOT,(uint32_t)500);		->		Futaba Tx(rpm sensor): 62
		send_SBS01RB(RPM_SLOT,(uint32_t)1000);		->		Futaba Tx(rpm sensor): 124
		send_SBS01RB(RPM_SLOT,(uint32_t)1000000);	->		Futaba Tx(rpm sensor): 49151

		send_RPM(RPM_SLOT,(uint32_t)500);		        ->		Futaba Tx(rpm sensor): 62
		send_SBS01RM(RPM_SLOT,(uint32_t)500);		->		Futaba Tx(rpm sensor): 62
		send_SBS01RO(RPM_SLOT,(uint32_t)500);		->		Futaba Tx(rpm sensor): 62
		send_SBS01R(RPM_SLOT,(uint32_t)500);		        ->		Futaba Tx(rpm sensor): 62

Here is a simple code just for testing

[Sbus2_write.zip](https://github.com/BrushlessPower/SBUS2-Telemetry/files/12802586/Sbus2_write.zip)

ericlangel, 2023-10-04:
T16 SZ and R7008SB should be fine
your sketch with Timer should be fine, too.

i'am wondering about the value 62

The RPM Value is divided by 6 -> transmitted via SBUS2 and Fasstest -> and multiplied by 6 on your T16SZ

so every RPM value is a factor of 6

so there is no way to transmit a value of 62 -> it should be 60 or 66

But the T16SZ RPM Sensor has additional settings
[image]

Can you check your Sensor Type and Gear Ratio?
i think this setting is "wrong"

ericlangel, 2023-10-04:
you can use both, but 3v3 is more easy going.

But with pro mini you need some electrical parts for inverting the SBUS Signal. if you just want to have a smaller MCU you could have a look on the RP2040 project

https://github.com/mstrens/oXs_on_RP2040

billis-xania, 2023-10-04:
Very nice project with the RP2040. 
Thank's again for your help.

ericlangel, 2023-10-05:
so i think i can close the issue

closed 2023-10-05 by ericlangel

ericlangel, 2023-10-05:
and is it working?

billis-xania, 2023-10-06:
Yes, it's work just fine. I'll send also the code if anyone want it.
[RPM_sensor_SBUS2_pro_mini.zip](https://github.com/BrushlessPower/SBUS2-Telemetry/files/12829572/RPM_sensor_SBUS2_pro_mini.zip)

jianwen0527, 2023-10-10:
Hello! The code you sent cannot be successfully edited, 
prompt:no matching function for call to 'SimpleTimer::setInterval(long int, void (&)())'

billis-xania, 2023-10-11:
Hi, can you try with the following library?
[SimpleTimer-master.zip](https://github.com/BrushlessPower/SBUS2-Telemetry/files/12865582/SimpleTimer-master.zip)

jianwen0527, 2023-10-12:
Thank you! Successfully compiled!