BrushlessPower/SBUS2-Telemetry #53: Why is there 2 voltages available for the SBS01T ?

Polohpi, 2023-10-24:
Hello

Great lib ! Works perfectly with a R304SB and a 4px. Just this question.

Thanks !

ericlangel, 2023-10-25:
Hello,
which is the first Voltage?
Which is the second Voltage?

The SBS01T is a Temperature Sensor.

Polohpi, 2023-10-25:
Yes my bad it is the SBS01V sorry. 

`send_voltage(VOLTAGE_SLOT,(uint16_t)128, (uint16_t)255);`  this send 2 voltages. The only one that i can read is the second one. I dont understand how the 01V can measure 2 inputs.

BrushlessPower, 2023-10-26:
the SBS01V is a Sensor with 2 Slots
one for the internal Voltage comming from RX (BEC Voltage)
One for the external input

I'am not sure about the R304SB. Does it send RX Voltage over Telemetry?

Maybe.....if RX Voltage (measured from RX) is different to internal Voltage send by SBS01V. There could be a Problem.
Maybe....there is a bug in Code.

I could check with my T14SG.
Did you tried another SLOT? Which Slot is set for SBS01V?

Polohpi, 2023-10-27:
The rs304b already send internal voltage on slot 0. That's maybe why. 
 
I dont think it's a bug, i have the same problem with the exemple included in the library.

BrushlessPower, 2023-10-27:
maybe my Code is buggy ;)

maybe the internal Voltage of RX (RS304b) and internal Voltage of SBS01V have to be the same value (or similar value).
I'll test it with T14SG