BrushlessPower/SBUS2-Telemetry #54: SBUS Channel data

BRolandG, 2023-11-21:
Hi
I'm building a RC plane tween engine sync device based on UM TINYPICCO. Using the engine's RPM values to adjust the slave engine's throttle position via a PID. I need both channel data and sensor data. Trottel, rudder, gain, and a switch channel to operate it. Sensors nnede are RPM for the master engine and the slave engine.

My problem is that the channel data received via the SBUS channel ranges from 352 to 1695. I expected it to range from 1101us to 1942us. I measured the expected values with my oscilloscope.

Bernt

ericlangel, 2023-11-21:
Hello,

it sounds ok to me

it should be 172 and 1811 for 1000µs and 2000µs

BRolandG, 2023-11-21:
Hi

Thanks for Your quick reply. I assumed the values (172 to 1811) to be in us format so I could send them directly to the slave engine servo from my Tinypicco. There is something I don't understand here....

Bernt

ericlangel, 2023-11-23:
Hello,

for a correct answer you have to ask Futaba.

but i guess that a Dual Rate of 150% in Transmitter will end in SBUS Values from 0 to 2000 (or 2048). and that would be around 900µs to 2100µs

As you said, the easiest way is map().

Could you try the 150% Servo in Transmitter and tell us the SBUS Values?

BRolandG, 2023-11-23:
Hi, yes still struggling with this nonlinear conversion.

I have used  T32MZ and R7008SB and Picoscope7 to measure pwm.

Bernt

Sbus value min 100% mid max 100%
1 352 1023 1695
2 354 1027 1695
3 352 1025 1695
4 352 1023 1695
Pwm values
1 1100 1520 1940
2 1100 1523 1940
3 1100 1522 1940
4 1100 1520 1940
3,125 1,486 1,145

BRolandG, 2023-11-23:
Hi again.

I was wrong it is linear.. sorry. Map works fine even with the transmitter
endpoints set to 150%.

Bernt

Sbus min endpoint max endpoint
1 84 1963
2 89 1963
3 84 1963
4 84 1963
PWM
1 933 2108
2 937 2108
3 933 2108
4 935 2108

ericlangel, 2023-11-24:
Thank you for sharing your results.

I would close the Issue at this point, because i cant change anything. Its a Futaba/SBUS specific think.

closed 2023-11-24 by ericlangel

BRolandG, 2023-11-24:
Hi

Yes problem solved. It works perfekt as it is. It was my misstanke.

Thanks!

Bernt

ericlangel, 2023-11-27:
Nice to Know

BRolandG, 2023-12-08:
Hi

Thanks for your nice code, it works like a charm.. I do have one question
for you.

I run this code in my loop and  poll for new data. I want the fresh frames
as soon as they are available, but I don't want to use that data unless it
is fresh. As i works with this code  the slots is ok but the SBUS_Ready
function is true for any frame. So I reuse the channel data as if it was
fresh.

The purpose of my code is to synchronize two engines RPM with a PID and I
want to feed the PID  ony with new data.

Bernt

   - SBUS_Ready() -> True when new Servo data is received with SBUS and
   SBUS2 Frames

  channelDataAvail = false;
  if (SBUS_Ready()){
    tpMaster = getChannel(MASTER_ENGINE_TROTTEL_CHANNEL);
    runningMode = getMode(getChannel(MODE_CHANNEL));      //mode 0
masterTP,mode 1 med PID,mode 2 PID och rudderoffset
    rudderPosition = getChannel(RUDDER_CHANNEL);
    gainPosition = getChannel(GAIN_CHANNEL);
    channelDataAvail = true;
  }
  masterRpmAvail = false;
  byte high = 0, low = 0;
  if (SBUS2_get_Slot(RPM_SLOT_MASTER, &low, &high)){
    rpmMaster = ((high << 8) | low) * 6;             //Ger RPM
    masterRpmAvail = true;
  }
  slaveRpmAvail = false;
  if (SBUS2_get_Slot(RPM_SLOT_SLAVE, &low, &high)){
    rpmSlave = ((high << 8) | low) * 6;             //Ger RPM
    slaveRpmAvail = true;
  }

ericlangel, 2023-12-11:
in the Master Branch its already fixed, but not uploaded in arduino Lib Manager

BRolandG, 2023-12-11:
Hi Yes I tested that, thanks.

However what I'm looking for is a way to poll for *just* new frames. I
guess that works fine for telemetry slots since the SBUS2_Ready() is true
every 56ms. But SBUS_Ready() is true every 14ms.

Bernt

BRolandG, 2023-12-11:
> normally sbus2_ready should also trigger every 14ms
>
> every SBUS(2) Frame contains complete Servo Channel data with an Update
> interval of 14ms.
>
> but complete Telemetry Data needs 4 SBUS(2) Frames. So Telemetry can be
> updated 4 x 12ms = 56ms.
>
> If i understand you correctly:
> you have two Motors and want to sync them. So you need RPM Value and
> Throttle Value very fast.
>
> What you could do:
> Make 4 RPM Sensor Slots with the same actual RPM Value
>
> Slot 1 (SBUS Frame 1)
> Slot 8 (SBUS Frame 2)
> Slot 16 (SBUS Frame 3)
> Slot 24 (SBUS Frame 4)
>
> So your actual RPM Value is send every 14ms.
> now you "just" have to read the SBUS2 Frame every 14ms

Hi thanks.

That is creative.. Good idea, I will try that.

Bernt

ericlangel, 2023-12-11:
not sure
i think you dont need your TX.

how is you setup?
one ESP for one engine?
so Engine 1 is master and engine 2 is slave?

and the Slave sends RPM Value to the Master via SBUS?

BRolandG, 2023-12-11:
My setup is

One airframe 2 engines.

Master engine trottel servo runs directly from RX.
Slave engine trottel servo runs directly from ESP.

I also need data from 4 channels in ESP.
Master trottel
Mode
Gain
Rudder

During startup I control slave trottel pwm with master trottel pwm. Rudder
is used to set an RPM offset to PID setpoint, based on mode and gain
channels.

The ESP is running the PID. The PID setpoint is master RPM, actual value is
slave RPM. The PID then deliver output as PWM trottelposition for the slave
engine. I PWM write that directly to the slave engines servo from the ESP.

Bernt

ericlangel, 2023-12-12:
so you are having just one esp in your plane?
or two?

i dont understand what is connectet to SBUS?

p.s. a little bit offtopic: watch out with PWM generated by ESP -> if the ESP have a firmwareissue the PWM is generated with the last value all the time, you cant stop that engine.

i'am doing somthing similar. i get the throttle value from RX and modify that value and send it to the ESC. two years ago i had the isuue that i couldnt stop because the esp was blocked somewere.

BRolandG, 2023-12-12:
Hi thanks.

They say that a picture/drawing is more clear than a 1000 words. Perhaps my
code will explain what I'm trying to do....

Bernt

ericlangel, 2023-12-12:
so you have one ESP and 2 Futaba RPM Sensors?

BRolandG, 2023-12-12:
Yes that is correct.

Bernt

ericlangel, 2023-12-12:
ok now i understand your hardware.

so with the Futaba RPM Sensor you will just receive the RPM Value every 56ms. You cant get it faster.

BRolandG, 2023-12-12:
Yes that is right.

I would really like to have a more frequent RPM update. The sensor detect a
signal every 6ms @ 10000 rpm.

Bernt

ericlangel, 2023-12-12:
if price doesnt matter, you could use 4 Futaba RPM Sensors.
Each sensor with its own Slot

Engine 1
Slot 1 (SBUS Frame 1)
Slot 8 (SBUS Frame 2)
Slot 16 (SBUS Frame 3)
Slot 24 (SBUS Frame 4)

Engine 2
Slot 2 (SBUS Frame 1)
Slot 9 (SBUS Frame 2)
Slot 17 (SBUS Frame 3)
Slot 25 (SBUS Frame 4)

or building your own RPM sensor