BrushlessPower/SBUS2-Telemetry #55: ESP32 S2 crash in static void IRAM_ATTR uart_intr_handle()

chrisDupuis, 2024-05-22:
Crash with ESP32 S2 mini.

(Code is running on ESP32)

Someone have test on ESP32 S2 ?
Any idea how to solve that ?

ericlangel, 2024-05-22:
i dont have a S2 for testing

interesting that it is running on ESP32

which ESP32 Hardware Core Version?
can u use the Exception decoder?

chrisDupuis, 2024-05-22:
S2 Mini V1.0.0, 
[https://www.amazon.fr/Hailege-Development-ESP32-S2FN4R2-MicroPython-Compatible/dp/B0BN1NY2KJ/ref=sr_1_1_sspa?dib=eyJ2IjoiMSJ9.KBzfCxg-Yi51IQaoc_cGg6BZQUKnDdNMJtP9zHpVHmlyNO4vVlCoMiYXQ_uuQHtUSDTzQ338wkmle0topAY1lmSlsrOxUnuHED2iKTbMgfmwA3ARa-cx2q_q5YOIeoU1Tn2v9cGgKSUnnxeQpLjk0_B0mOgrMEhboOfeG4rDj4hXvylcoGeZ8ZjGE7XbddCwgXurVZrUMxVUnwlUmOBu21PbszNOI2RJXgXtFBmUwRLPIcaSQee5ERrMSkmpSuGwb2dWuh4er-ss8Z_8866vsTjCJqS1MXmV3jTlQMcPW-Y.ejP6gC0mcycPYIqJx8kgqVHaWbD1MQtkPVzwx7Zn3ZE&dib_tag=se&keywords=esp32+s2&qid=1716370217&sr=8-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1](url)

I have no exception shown, 
only constantly rebooting...

ericlangel, 2024-05-22:
I need the version of your ESP32 arduino core version
is it already 3.0.0?
or 2.0.16?
or?

ericlangel, 2024-05-22:
did you had a look at #44 ?

there are several reasons for crashing. It's hard to get the right one without exception decoder
i could try to get a S2 Board, but i need some time to rework the code

there are several known issues, but i had no time for fix

chrisDupuis, 2024-05-22:
Yes I use this topic and done some test since 3 days.

If I remove 
`ESP_ERROR_CHECK(uart_isr_register(uart_num, uart_intr_handle, NULL, ESP_INTR_FLAG_IRAM, 0 /*handle_console*/));`
No crash.

If I return in ISR uart, then crash.

static void IRAM_ATTR uart_intr_handle(void *arg)
{
  uint16_t status = UART1.int_st.val; // read UART interrupt Status
  // status = UART1.int_st.val;             // read UART interrupt Status
  uart_clear_intr_status(uart_num, UART_RXFIFO_FULL_INT_CLR | UART_RXFIFO_TOUT_INT_CLR);
  uart_clear_intr_status(uart_num, status);

  return;
}

ericlangel, 2024-05-22:
ESP_ERROR_CHECK(uart_isr_register(uart_num, uart_intr_handle, NULL, ESP_INTR_FLAG_IRAM, 0 /*handle_console*/));

this call register the function uart_intr_handle to the Interrupt
without this, the handler is never called so of course there is no Crash

you could try to register the function without ESP_INTR_FLAG_IRAM

chrisDupuis, 2024-05-22:
Same crash without ESP_INTR_FLAG_IRAM.

I try to have very simple uart_intr_handle with minimum code,
in ordre to not have crash/reboot.

chrisDupuis, 2024-05-22:
What about that for minimum code
`
static void IRAM_ATTR uart_intr_handle(void *arg)
// static void uart_intr_handle(void *arg)
{
  uart_clear_intr_status(uart_num, UART_RXFIFO_FULL_INT_CLR | UART_RXFIFO_TOUT_INT_CLR);
  return;
}
`

chrisDupuis, 2024-05-22:
One more, info,
If no SBUS2 signal, no crash.

ericlangel, 2024-05-23:
Ok lets try to work it out.

- No SBUS Signal no crash. Thats right. without SBUS signal the UART Interrupt is never executed

please uncomment the following lines/functions one by one and report back

- disable_receiving(); line321
- start_transmit_sequencer(((rxbuf[24]&0x30) >> 4)); line 330
- pinMatrixOutDetach(tx_pin,false,true); line265

the first 2 are working with the Timer, and i'am sure it is more or less a Timer Problem, but who knows

chrisDupuis, 2024-05-23:
Test comment this line:

disable_receiving(); line321                                                              => reboot
start_transmit_sequencer(((rxbuf[24]&0x30) >> 4)); line 330   => reboot
pinMatrixOutDetach(tx_pin,false,true); line265                             => reboot

chrisDupuis, 2024-05-23:
On more info, 
with the same module and pin assignment, I have test with iBus telemetry, and it s running ok
https://github.com/bmellink/IBusBM

So for me the hardware, and wire is ok.

chrisDupuis, 2024-05-23:
I add a print_reset_reason, and the cause is SW_RESET, RESET_REASON = 3

ericlangel, 2024-05-23:
> Test comment this line:
> 
> disable_receiving(); line321 => reboot start_transmit_sequencer(((rxbuf[24]&0x30) >> 4)); line 330 => reboot pinMatrixOutDetach(tx_pin,false,true); line265 => reboot

And all 3 disabled together?

ericlangel, 2024-05-23:
here is the answer: [Link](https://esp32.com/viewtopic.php?t=16723)

in the last comment

could you try to change:
rxbuf[buffer_index] = UART1.fifo.rw_byte; // read all bytes

to

rxbuf[buffer_index] = READ_PERI_REG(UART_FIFO_AHB_REG(1)); // read all bytes

But there seems to be a bigger issue with the UART Periph on S2. So maybe its not crashing any more but UART is not read correctly

chrisDupuis, 2024-05-23:
rxbuf[buffer_index] = READ_PERI_REG(UART_FIFO_AHB_REG(1)); // read all bytes
Same

chrisDupuis, 2024-05-23:
OK, the problem is here:
```
while (rx_fifo_len)
  {
    // CHD READ_PERI_REG(UART_FIFO_AHB_REG(0))
    // https://esp32.com/viewtopic.php?t=16723
    // CHD 
    // rxbuf[buffer_index] = UART1.fifo.rw_byte; // read all bytes
    //rxbuf[buffer_index] = UART1.ahb_fifo.rw_byte; // read all bytes
    rxbuf[buffer_index] = READ_PERI_REG(UART_FIFO_AHB_REG(1));

    rx_fifo_len--;
    buffer_index++;
    // Serial.print(rxbuf[buffer_index],HEX);
    // Serial.print(" ");
    /*if(rxbuf[0] == 0x0f){						// Start of SBUS(2) Frame
      buffer_index++;
    }
    else{										// Telemetry Slot?
    //digitalWrite(16,HIGH);
    buffer_index++;
    }*/
  }
```

chrisDupuis, 2024-05-23:
not found solution yet

chrisDupuis, 2024-05-23:
OK Running

ericlangel, 2024-05-24:
Nice to hear.

What did u change?
just `READ_PERI_REG(UART_FIFO_AHB_REG(1))` ?

chrisDupuis, 2024-05-26:
Here the code for ESP32 and ESP32_S2

```
/*
This program is free software: you can redistribute it and/or modify
it under the terms of the GNU General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.

You should have received a copy of the GNU General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/
#if defined(ESP32)
#include "SBUS_usart.h"
#include "esp32-hal.h"
#include "esp32-hal-uart.h"

// #include "freertos/FreeRTOS.h"
// #include "freertos/event_groups.h"
// #include "esp_task.h"
// #include "esp_task_wdt.h"
// #include "freertos/queue.h"
#include "driver/uart.h"
#include <soc/uart_reg.h>
#include <driver/timer.h>
#include <soc/uart_struct.h>
#include <soc/timer_group_struct.h>

#include <stdarg.h>
#include <stdlib.h>
#include <string.h>

#define SLOT_DATA_LENGTH 3
#define NUMBER_OF_FRAMES 4
#define NUMBER_OF_SLOT 32
#define NUMBER_OF_SLOT_IN_FRAME 8
#define SBUS_FRAME_SIZE 25

#if CONFIG_IDF_TARGET_ESP32
#define SBUS2_UART_RX_PIN (GPIO_NUM_25)
#define SBUS2_UART_TX_PIN (GPIO_NUM_26)
#elif CONFIG_IDF_TARGET_ESP32S2
#define SBUS2_UART_RX_PIN (16)
#define SBUS2_UART_TX_PIN (17)
#endif

#define UART_RXBUFSIZE 30

#define SLOT_TIME 660  // 660 ms
#define WAIT_TIME 2000 // 2000 ms

#define RX_BUF_SIZE 1024 // we dont need more space in buffer

bool DemoMode = false;

// 32 Slots for telemetrie Data
uint8_t Slot_ID[NUMBER_OF_SLOT] = {0x03, 0x83, 0x43, 0xC3, 0x23, 0xA3, 0x63, 0xE3,
                                   0x13, 0x93, 0x53, 0xD3, 0x33, 0xB3, 0x73, 0xF3,
                                   0x0B, 0x8B, 0x4B, 0xCB, 0x2B, 0xAB, 0x6B, 0xEB,
                                   0x1B, 0x9B, 0x5B, 0xDB, 0x3B, 0xBB, 0x7B, 0xFB};

char sbus_frame[25] = {0x0F, 0xA0, 0xA3, 0x20, 0x56, 0x2C, 0x08, 0x16, 0x50, 0x03, 0x10, 0x80, 0x00, 0x04, 0x20, 0x00, 0x01, 0x08, 0x07, 0xC8, 0x03, 0x10, 0x80, 0x02, 0x04};

// channels for connected servos at the SBUS
uint16_t channels[NUMBER_OF_CHANNELS];
[... 630 lines cut ...]

void SBUS_disable()
{
  uart_disable_rx_intr(uart_num);
  timer_set_alarm(TIMER_GROUP_0, TIMER_1, TIMER_ALARM_DIS);
}

void SBUS_enable()
{
  uart_enable_rx_intr(uart_num);
  initialize_slot_status();
  enable_receiving();
}

void SBUS2_disable()
{
  SBUS_disable();
}

void SBUS2_enable()
{
  SBUS_enable();
}
#endif // ESP32
```

chrisDupuis, 2024-05-26:
Fixed

ericlangel, 2024-05-27:
OK, i have some questions:

u used PIN16/17 on the S2 -> why? are 25/26 not working?

and u changed:

```
#if CONFIG_IDF_TARGET_ESP32
  TIMERG0.int_clr_timers.t1 = 1;
  TIMERG0.hw_timer[1].config.alarm_en = 1;
#elif CONFIG_IDF_TARGET_ESP32S2
  timer_group_clr_intr_status_in_isr(TIMER_GROUP_0, TIMER_1);
  TIMERG0.hw_timer[0].config.tx_alarm_en = 1;
#endif
```

`TIMERG0.hw_timer[0].config.tx_alarm_en = 1;` -> seems wrong to me
should it be` TIMERG0.hw_timer[1].config.tx_alarm_en = 1; `?

chrisDupuis, 2024-05-27:
I use 25/26 because I use an ESP32 S2 Mini Lolin, and no 25/26 pin.
and
You re right, it s a mistake, it must be:
  TIMERG0.hw_timer[1].config.tx_alarm_en = 1;