BrushlessPower/SBUS2-Telemetry #56: New version of UM TinyPico ESP32 system

BRolandG, 2024-06-22:
Hi

I just updated my the IDE
C:\Users\bernt\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32\Arduino.h

And I'm getting this... can you help?

In file included from C:\Users\bernt\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/esp_hw_support/include/esp_intr_alloc.h:11,
                 from C:\Users\bernt\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/esp_hw_support/include/esp_cpu.h:20,
                 from C:\Users\bernt\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/esp_hw_support/include/spinlock.h:11,
                 from C:\Users\bernt\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:74,
                 from C:\Users\bernt\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/freertos/FreeRTOS-Kernel/include/freertos/portable.h:59,
                 from C:\Users\bernt\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:71,
                 from C:\Users\bernt\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/Arduino.h:33,
                 from c:\Users\bernt\OneDrive\Dokument\Arduino\libraries\SBUS2-Telemetry-master\src\SBUS_usart.h:15,
                 from c:\Users\bernt\OneDrive\Dokument\Arduino\libraries\SBUS2-Telemetry-master\src\SBUS_usart_ESP32.cpp:16:
c:\Users\bernt\OneDrive\Dokument\Arduino\libraries\SBUS2-Telemetry-master\src\SBUS_usart_ESP32.cpp: In function 'void SBUS2_uart_setup(int, int, int)':
c:\Users\bernt\OneDrive\Dokument\Arduino\libraries\SBUS2-Telemetry-master\src\SBUS_usart_ESP32.cpp:173:52: error: 'UART_INVERSE_RXD' was not declared in this scope
  173 |    ESP_ERROR_CHECK(uart_set_line_inverse(uart_num, UART_INVERSE_RXD | UART_INVERSE_TXD));                 // V1.6.0
      |                                                    ^~~~~~~~~~~~~~~~
C:\Users\bernt\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/esp_common/include/esp_err.h:117:30: note: in definition of macro 'ESP_ERROR_CHECK'
  117 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
c:\Users\bernt\OneDrive\Dokument\Arduino\libraries\SBUS2-Telemetry-master\src\SBUS_usart_ESP32.cpp:173:71: error: 'UART_INVERSE_TXD' was not declared in this scope
  173 |    ESP_ERROR_CHECK(uart_set_line_inverse(uart_num, UART_INVERSE_RXD | UART_INVERSE_TXD));                 // V1.6.0
      |                                                                       ^~~~~~~~~~~~~~~~
C:\Users\bernt\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/esp_common/include/esp_err.h:117:30: note: in definition of macro 'ESP_ERROR_CHECK'
  117 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
c:\Users\bernt\OneDrive\Dokument\Arduino\libraries\SBUS2-Telemetry-master\src\SBUS_usart_ESP32.cpp:176:4: error: 'gpio_set_pull_mode' was not declared in this scope; did you mean 'gpio_pull_mode_t'?
  176 |    gpio_set_pull_mode((gpio_num_t)rx, GPIO_PULLDOWN_ONLY);
      |    ^~~~~~~~~~~~~~~~~~
      |    gpio_pull_mode_t
c:\Users\bernt\OneDrive\Dokument\Arduino\libraries\SBUS2-Telemetry-master\src\SBUS_usart_ESP32.cpp:177:4: error: 'gpio_pulldown_en' was not declared in this scope; did you mean 'gpio_pulldown_t'?
  177 |    gpio_pulldown_en((gpio_num_t)rx);
      |    ^~~~~~~~~~~~~~~~
      |    gpio_pulldown_t
c:\Users\bernt\OneDrive\Dokument\Arduino\libraries\SBUS2-Telemetry-master\src\SBUS_usart_ESP32.cpp:179:20: error: 'uart_isr_free' was not declared in this scope; did you mean 'uart_lost_frame'?
  179 |    ESP_ERROR_CHECK(uart_isr_free(uart_num));
      |                    ^~~~~~~~~~~~~
C:\Users\bernt\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/esp_common/include/esp_err.h:117:30: note: in definition of macro 'ESP_ERROR_CHECK'
  117 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
c:\Users\bernt\OneDrive\Dokument\Arduino\libraries\SBUS2-Telemetry-master\src\SBUS_usart_ESP32.cpp:180:20: error: 'uart_isr_register' was not declared in this scope; did you mean 'timer_isr_register'?
  180 |    ESP_ERROR_CHECK(uart_isr_register(uart_num,uart_intr_handle, NULL, ESP_INTR_FLAG_IRAM, 0/*handle_console*/));
      |                    ^~~~~~~~~~~~~~~~~
C:\Users\bernt\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32/include/esp_common/include/esp_err.h:117:30: note: in definition of macro 'ESP_ERROR_CHECK'
  117 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
c:\Users\bernt\OneDrive\Dokument\Arduino\libraries\SBUS2-Telemetry-master\src\SBUS_usart_ESP32.cpp: In function 'void ISR_transmit_frame(void*)':
c:\Users\bernt\OneDrive\Dokument\Arduino\libraries\SBUS2-Telemetry-master\src\SBUS_usart_ESP32.cpp:388:26: error: 'volatile union timg_int_clr_timers_reg_t' has no member named 't1'
  388 |   TIMERG0.int_clr_timers.t1 = 1;
      |                          ^~
c:\Users\bernt\OneDrive\Dokument\Arduino\libraries\SBUS2-Telemetry-master\src\SBUS_usart_ESP32.cpp:389:30: error: 'volatile union timg_txconfig_reg_t' has no member named 'alarm_en'
  389 |   TIMERG0.hw_timer[1].config.alarm_en = 1;
      |                              ^~~~~~~~

exit status 1

Compilation error: exit status 1

ericlangel, 2024-06-24:
dear @BRolandG 

The new ESP32 V3.0.0 Core has made a big jump in its API calls and Register names.
Actual, there is no support for it.
Even the V2.0.17 already made some problems. 

i have an own ESP32 project for RC Apllication. And i'am still on V2.0.17.
This project need's a complete clean up. 
I hope to get it working in the next weeks.

BRolandG, 2024-07-08:
Hi

Ok. What version would you say is last stabile? 2.0.16??

Bernt

ericlangel, 2024-07-09:
2.0.17 should be ok with normal esp32