BrushlessPower/SBUS2-Telemetry #57: JetCat's status slot can't work

kevinuav, 2024-08-04:
I have already successfully sent the RPM,EGT on jetcat's slot. But the status. Why? Is there something preprocessing I miss before the status code sent into the "SBUS2_transmit_telemetry_data()"
here is the RPM,EGT:
[image]
And here is the status interface:
[image]

ericlangel, 2024-08-12:
Hello,

the Problem with the JetCat Sensor is, that my T14SG doesnt have this Sensor. So i have to code it blind/without testing

The Status is actual a Value from 0 to 65535.

But i have no idea which Value represent which Status.
can your try the following values and tell me if status is changed and which status?

0x0000
0x0001
0x0002
0x0004
0x0008
0x0010
0x0020
0x0040
0x0080
0x0100
0x0200
0x0400
0x0800
0x1000
0x2000
0x4000
0x8000

this could work for 16 different status Flags. In your picture i can see 9 different Status flags.
but maybe Futaba or Jetcat did something special.

kevinuav, 2024-08-12:
So do you mean the status code doesn't correspond to the ECU's state code/off-condition code?

ericlangel, 2024-08-12:
I would guess that the Off condition is send in the Status Slot.
But i cant know it.

Which Status values did you test?

my values above could bei similar to:

0x0000 -> Everything Off/ Error
0x0001- >Temp OK
0x0002-> Battery OK
0x0004 -> Sensor OK
0x0008 -> Off Condition
and so on

But, we (you) need to test it. Or ask Futaba or Jetcat.

ericlangel, 2024-08-13:
> I have asked jetcat's distributor. They say the sbus2 code is corresponding to the ECU's status code.

Can you send me the Jetcat Codes?

> BTW. Why don't you buy a new futaba 18sz while you have a jetcat? : )

I have no Jetcat

kevinuav, 2024-08-13:
I can just send it to you personally. But in this public.

ericlangel, 2024-08-13:
send it to:
brushlesspower@protonmail.com

kevinuav, 2024-08-13:
OK

kevinuav, 2024-08-13:
> send it to: [brushlesspower@protonmail.com](mailto:brushlesspower@protonmail.com)

So you can see the status codes max value is only 19, it is not reasonable to send it with a 16 bits slot. It is waste of data resource.

ericlangel, 2024-08-13:
ah yeah, i see.

try a value between 16384 to 16404
maybe its singned integer value.

16384 == State 0
16404 == State 19

kevinuav, 2024-08-13:
> ah yeah, i see.
> 
> try a value between 16384 to 16404 maybe its signed integer value.
> 
> 16384 == State 0 16404 == State 19

Tried it doesn't work.

kevinuav, 2024-08-13:
First of all. Are you sure the 12rd slot is for the status code?

ericlangel, 2024-08-13:
yes and no
someone @i3dm decoded the Jetcat SBUS2 Sensor.
He was reading all sensor data and figured out which Slot is which value.

Of course there could be a mistake in his findings or my implementation.

kevinuav, 2024-08-22:
> any news?

No，I have tried many  value with different data type.  It always shows 2 bars in the status.

kevinuav, 2024-09-02:
> Hello,
>
> the Problem with the JetCat Sensor is, that my T14SG doesnt have this Sensor. So i have to code it blind/without testing
>
> The Status is actual a Value from 0 to 65535.
>
> But i have no idea [...]

Hi，I have tried many test and they all fail. But someone says your T14SG should work with jetcat after you update the firmware of the remote controller. Why don't you have a try?

ericlangel, 2024-09-02:
Hi,

good question.
1. my T14SG is V4.1 without LBT. The system is running without any Problems......so never change running system
2. i could Update my second T14SG...i think its already with LBT, but in every Chart its pointed that the T14SG has no support for the JetCat Sensor. So i never had the need to update.
3. i have no jetcat at all. Your are the first who is using this sensor (or having problems)