BrushlessPower/SBUS2-Telemetry #58: FrameLength issue

BRolandG, 2024-08-29:
Hi

Something is wrong here.

I'm trying to sync 2 engines rpm with ESP32 (v2.0.17).  Futaba T32MZ and R7008SB, only receiving channels and slots.

The display in the logg comes from Serial.print in the cpp file. I just removed the /*.

I have WDT set to panic after 1 sek.

Bernt

[logg.txt](https://github.com/user-attachments/files/16795693/logg.txt)
[TweenEngineSyncSbus2.zip](https://github.com/user-attachments/files/16795815/TweenEngineSyncSbus2.zip)

ericlangel, 2024-08-29:
Hi Bernt,

i know we already talked about your project.
Could you just share in some words what you are trying?

2 engines -> each engine has its own original futaba rpm sensor? (or more than one?)
and you are controlling the slave engine from ESP32 based on the received RPM Values?

anything else attached on the SBUS2?

BRolandG, 2024-08-29:
Hi

I’m using 5 Futaba sensors 2 RPM 2 for temp and one for airspeed (+
external power on 0). All servos but ignition and gear are run through the
SBUS2 outlet on the receiver.

Feeding a PID with the master engine RPM as setpoint and the slave engine
RPM as input, PID then calculates the required PWM signal that I feed the
slave engine servo via the AnalogWrite software.

https://github.com/Dlloydev/ESP32-ESP32S2-AnalogWrite
https://github.com/imax9000/Arduino-PID-Library

Bernt

BRolandG, 2024-08-29:
Running the log through ExceptionDecoder gives this

PC: 0x4008cb76: vListInsert at
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos\list.c:183
EXCVADDR: 0x00000000

Decoding stack results
0x4008cb73: vListInsert at
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos\list.c:183

ericlangel, 2024-08-29:
ok, now we should break some things down:

are the 5 Sensors are working (without the ESP32) and you can see them with correct values in your T32MZ?
Which Slots are they using?

Are the Servos SBUS or SBUS2 types? i think there exist SBUS2 Servos which act like a sensor.....bur i'am not sure.

can you make a Quick & dirty esp32 sketch which is just reading the Sensor values? For example my Read Telemetry example?

BRolandG, 2024-08-29:
Hi

Yes the 5 sensors work with or without the ESP connected. They are using
slot 1-5.

All servos are SBUS2.

If I only read slots (no channels) then the sensor values are ok both on
T32MZ and MC.

Sketch.. perhaps this will do? I have one SBUS2 outlet on the receiver, it
is connected to the MC and a (futaba 6 way) hub (Y cable) then the
remainder of servos and sensors are linked together via that hub.
Then I use channel 9 and 10 for ignition and gear, connected to outlet 1
and 2 on the recceiver.

Bernt

ericlangel, 2024-08-29:
ok, at which point does the esp crash?

Error! -> rx_fifo_len = 6 Value = 83, DB, 2, 1E, D4, FE, 27,  [HEX]
Error! -> rx_fifo_len = 29 Value = 83, DB, 2, 1E, D4, FE, 27, BA, 6D, 84, 85, B1, A0, 6, B, 63, 1, 8, 40, 0, 82, 5, 80, 0, 4, 0, 0, 0, 0, 0,

these Errors are not nice, but it can happen.
if it is happen every 1000 Frames....doesnt matter
it its happen every 100 Frames.....ok not nice
if its happen every 10 Frames....you should worry

but it should never cause a crash.

so please tell me when does it happen and how often?
and....you shouldnt do Serial print in the UART handler (without knowing waht you do)...it could cause the crash.

BRolandG, 2024-08-29:
Hi

Ok understand.

It doesn’t crash I restart it if the main loop doesn’t get a run for 1
Sekond.  It seems to hang displaying Wrong FrameLength until my WDT restart
it.  On occasion I get just a couple of displays and then it is running ok
again, that is fine but not the hang thing.

The displays in the uart handler was just to have a look.

I use the package as is.

Bernt

ericlangel, 2024-08-29:
oh......Ok.

i think i got it.

So maybe some explaining:
the UART handler is executed when 25 Byte or more are in UART buffer (after a Frame) or after a timeout without new received bytes (after a sensor Slot with 3 bytes).

if there are not 25 or 3 Bytes in the UART buffer -> something is wrong and UART buffer needs to be cleared.
this should be done by `uart_flush(uart_num);`

1. thing
dont Serial.print "Error! -> rx_fifo_len = 6 Value = 83, DB, 2, 1E, D4, FE, 27,  [HEX]"
it's based on rxbuf which is 30Bytes big. if you try to print byte 31 something wired will happen

2. thing
we should clear the UART buffer manual right before uart.flush()
`while(rx_fifo_len){
    rxbuf[buffer_index] = UART1.fifo.rw_byte; // read all bytes
    rx_fifo_len--;
	buffer_index++;
  }`
  
  this should solve the problem

BRolandG, 2024-08-29:
Thanks, I will test.

Like this then?
 else{
 //digitalWrite(18,HIGH);
 /*Serial.print("Error! -> rx_fifo_len = ");
 Serial.print(rx_fifo_len);
 Serial.print(" Value = ");
 for(uint8_t i=0; i<=rx_fifo_len;i++){
Serial.print(rxbuf[i], HEX);
Serial.print(", ");
 }
 Serial.println(" [HEX]");*/
 while(rx_fifo_len){
rxbuf[buffer_index] = UART1.fifo.rw_byte; // read all bytes
rx_fifo_len--;
buffer_index++;
 }
 uart_flush(uart_num);
 uart_clear_intr_status(uart_num,
UART_RXFIFO_FULL_INT_CLR|UART_RXFIFO_TOUT_INT_CLR);
 log_i("[SBUS2] Wrong Frame length\l\n");
 return;
  }

ericlangel, 2024-08-30:
yes, this should work