BrushlessPower/SBUS2-Telemetry #61: can't compile for any ESP32 boards

Riktor1990, 2025-04-22:
I'm using Arduino ide and esp32-c6 but also tried c2 and c3. I have error from the library. Can anybody help me ?

ericlangel, 2025-04-22:
Hello,

could you tell me the ESP32 Core Version?
There are Problems with Core Version 3.0.0 and newer

Riktor1990, 2025-04-22:
It has automatically installed 3.2

ericlangel, 2025-04-22:
all errors are coming from the INO file. nothing from the library code itself.

Try it with the example code "SBUS2 Telemetry"

Riktor1990, 2025-04-22:
I simply put this code: `#include <SBUS2.h>

void setup() {
  Serial.begin(9600);
  Serial.println("SBUS2 library works!");
}

void loop() {}`

and I got this results:

> c:\Users\xxxxxxxxxx\Documents\Arduino\libraries\SBUS2\src\SBUS_usart_ESP32.cpp:126:5: error: cannot convert 'bool' to 'timer_alarm_t' in initialization
  126 |     };
      |     ^
In file included from C:\Users\xxxxxxxxxx\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32c6/include/esp_hw_support/include/esp_intr_alloc.h:12,
                 from C:\Users\xxxxxxxxxx\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32c6/include/esp_hw_support/include/esp_cpu.h:20,
                 from C:\Users\xxxxxxxxxxxx\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32c6/include/esp_hw_support/include/spinlock.h:11,
                 from C:\Users\xxxxxxxxxxx\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32c6/include/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos/portmacro.h:65,
                 from C:\Users\xxxxxxxxxxx\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32c6/include/freertos/FreeRTOS-Kernel/include/freertos/portable.h:57,
                 from C:\Users\xxxxxxxxxxx\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32c6/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:69,
                 from C:\Users\xxxxxxxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32/Arduino.h:33,
                 from c:\Users\xxxxxxxxxxx\Documents\Arduino\libraries\SBUS2\src\SBUS_usart.h:15,
                 from c:\Users\xxxxxxxxxxx\Documents\Arduino\libraries\SBUS2\src\SBUS_usart_ESP32.cpp:16:
c:\Users\xxxxxxxxxxx\Documents\Arduino\libraries\SBUS2\src\SBUS_usart_ESP32.cpp: In function 'void SBUS2_uart_setup()':
c:\Users\xxxxxxxxxxx\Documents\Arduino\libraries\SBUS2\src\SBUS_usart_ESP32.cpp:139:54: error: 'UART_INVERSE_RXD' was not declared in this scope
  139 |    ESP_ERROR_CHECK(uart_set_line_inverse(UART_NUM_1, UART_INVERSE_RXD|UART_INVERSE_TXD));
      |                                                      ^~~~~~~~~~~~~~~~
C:\Users\xxxxxxxxxxx\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32c6/include/esp_common/include/esp_err.h:104:30: note: in definition of macro 'ESP_ERROR_CHECK'
  104 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
c:\Users\xxxxxxxxxxx\Documents\Arduino\libraries\SBUS2\src\SBUS_usart_ESP32.cpp:139:71: error: 'UART_INVERSE_TXD' was not declared in this scope
  139 |    ESP_ERROR_CHECK(uart_set_line_inverse(UART_NUM_1, UART_INVERSE_RXD|UART_INVERSE_TXD));
      |                                                                       ^~~~~~~~~~~~~~~~
C:\Users\xxxxxxxxxxx\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32c6/include/esp_common/include/esp_err.h:104:30: note: in definition of macro 'ESP_ERROR_CHECK'
  104 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
c:\Users\xxxxxxxxxxx\Documents\Arduino\libraries\SBUS2\src\SBUS_usart_ESP32.cpp:144:20: error: 'uart_isr_free' was not declared in this scope; did you mean 'uart_lost_frame'?
  144 |    ESP_ERROR_CHECK(uart_isr_free(UART_NUM_1));
      |                    ^~~~~~~~~~~~~
C:\Users\xxxxxxxxxxx\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32c6/include/esp_common/include/esp_err.h:104:30: note: in definition of macro 'ESP_ERROR_CHECK'
  104 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
c:\Users\xxxxxxxxxxx\Documents\Arduino\libraries\SBUS2\src\SBUS_usart_ESP32.cpp:145:20: error: 'uart_isr_register' was not declared in this scope; did you mean 'gpio_isr_register'?
  145 |    ESP_ERROR_CHECK(uart_isr_register(UART_NUM_1,uart_intr_handle, NULL, ESP_INTR_FLAG_IRAM, 0/*handle_console*/));
      |                    ^~~~~~~~~~~~~~~~~
C:\Users\xxxxxxxxxxx\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32c6/include/esp_common/include/esp_err.h:104:30: note: in definition of macro 'ESP_ERROR_CHECK'
  104 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
c:\Users\xxxxxxxxxxx\Documents\Arduino\libraries\SBUS2\src\SBUS_usart_ESP32.cpp:149:31: error: 'TIMER_1' was not declared in this scope; did you mean 'TIMER_0'?
  149 |     timer_init(TIMER_GROUP_0, TIMER_1, &timer_config);
      |                               ^~~~~~~
      |                               TIMER_0
c:\Users\xxxxxxxxxxx\Documents\Arduino\libraries\SBUS2\src\SBUS_usart_ESP32.cpp:159:34: error: 'UART_RXFIFO_FULL_INT_ENA_M' was not declared in this scope
  159 |     uart_intr.intr_enable_mask = UART_RXFIFO_FULL_INT_ENA_M | UART_RXFIFO_TOUT_INT_ENA_M;
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\xxxxxxxxxxx\Documents\Arduino\libraries\SBUS2\src\SBUS_usart_ESP32.cpp:159:63: error: 'UART_RXFIFO_TOUT_INT_ENA_M' was not declared in this scope
[... cut ...]
      |   TIMER_0
c:\Users\xxxxxxxxxxx\Documents\Arduino\libraries\SBUS2\src\SBUS_usart_ESP32.cpp:288:42: error: 'TIMER_1' was not declared in this scope; did you mean 'TIMER_0'?
  288 |     timer_set_alarm_value(TIMER_GROUP_0, TIMER_1, SLOT_TIME);
      |                                          ^~~~~~~
      |                                          TIMER_0
Using library SBUS2 at version 1.2.0 in folder: C:\Users\xxxxxxxxxxx\Documents\Arduino\libraries\SBUS2 
exit status 1

Compilation error: exit status 1

ericlangel, 2025-04-22:
i dont know why it is not working
there are people who have got it working/compiled with ESP32-S2

i have to spend some time to make it working again. But in the least years i had no time do do it

Riktor1990, 2025-04-23:
Would be nice to update compatibility with core 3.0

ericlangel, 2025-05-12:
no, sorry

luisgarag, 2025-09-17:
I would really appreciate compatibility with core 3.0 too. Hope it can be updated!

ericlangel, 2025-09-17:
Just had a look on ESP32 core
we are already at V3.3.1....damn time is running.

Let me try it in the next days.
Is anybody already working on it?

ericlangel, 2025-10-22:
Hey,
i just had a short look. Interesting that it is working with the UART Driver.
I played a lot with LL functions to get around the ESP Drivers.

For me i have the problem that some direct Register access functions are available on one chip series, but not on the other.
So i have to use LL functions which should take care of all Chip series now and for future use.

Or i should try it with your solution.